tompi/cheapino #1: Plans to ENC1 re-route.

azhizhinov, 2023-04-03:
Hello.
Thanks for your time.
Have you any intentions to re-route encoder a bit?
Is it possible to use GP7 and GP8 for A and B pins?
[image]

tompi, 2023-04-03:
Hi. Thanks for taking a look! 

The encoder is on the opposite side of the mcu, and sides are connected using rj45, meaning there are only 8 wires available. 

I need 7 of these for the matrix(18 + 1 switches) so the encoder can’t really have 2 pins for itself…

If you check my qmk fork, I have it working fine the way it’s wired now

azhizhinov, 2023-04-03:
Oh, you're right.
Sorry.
So tricky approach.

closed 2023-04-03 by azhizhinov