tompi/cheapino #2: Hotswap support

finex, 2023-04-06:
Hi, the project is super interesting. Did you plan to add the hotswap feature in the future and to release the gerber file?

DeBasti, 2023-04-10:
Regarding the gerber files: You can install kicad, open the project and plot the gerber & drill files yourself to then upload it to your favorite pcb manufacturer.
I just did this with JLCPCB. They have a FAQ where they tell you how to configure kicad to get proper gerber files so that they can process them. Not sure if it works properly cause it's my first time doing this but I'm quite confident.

I also had a look at pcbway and they need different settings for the plotting of the gerber files. So I think that might be why he/she/them did not include them in the repository (just guessing).

tompi, 2023-04-14:
hey, i uploaded the gerber.zip as release 1.0. Keeping the issue for reference if I make hotswaps for next version.

finex, 2023-04-14:
Thank you very much! I really appreciate it!  :-)

tompi, 2024-04-06:
Fixed in 1355b80bf61a885dfb9796597a6b3016640277ef

closed 2024-04-06 by tompi