tompi/cheapino #3: Support 5 pin switch

axhixh, 2023-04-06:
Using 5 pin switch will give a more stable, wobble free keyboard without using a plate. Please change the switch footprint to support it.

tompi, 2023-04-14:
hey, will try to fix it for next release, will try to use a footprint like this: https://github.com/daprice/keyswitches.pretty/blob/master/images/Kailh_socket_MX_reversible.png

DeBasti, 2023-04-26:
Would be really nice if the footprint could also support choc v1 switches!

axhixh, 2023-05-18:
Does that foot print support using it without hot swap socket? It doesn’t look like it does.

tompi, 2023-05-18:
Oops, this should probably work: 
[image]

keyboard-magpie, 2023-09-16:
[cheapino plate.stl.zip](https://github.com/tompi/cheapino/files/12641479/cheapino.plate.stl.zip)
Made a basic plate for a user on the lowprofilekb server.

zawnk, 2023-10-04:
@tompi the plate.stl that you uploaded to the repo doesn't seem to work properly. I tried looking at it at https://www.viewstl.com/ or inside Blender and it didn't work, when I loaded it into JLCPCB it error'd out, too. With the direct download and unzipping from the comment above, it works just fine (your plate.stl is also like 350kb and the unzipped stl from the comment is 100kb).

tompi, 2024-04-06:
Fixed in 1355b80bf61a885dfb9796597a6b3016640277ef

closed 2024-04-06 by tompi