tompi/cheapino #4: Share (affiliate?) links

Throvn, 2023-04-25:
I really like your build. I want to get my own split keyboard but all of the assembled solutions are so god damn expensive. Unfortunately, I'm still a noob when it comes to hardware electronics. Things like: `diodes (both smb and throug hole are supported)` don't make much sense to me. That's why you would help me a lot when you would share the links to the products you ordered somewhere. I don't think anyone would mind if these are affiliate links, but that's up to you.

Thank you for the time. I really appreciate your work!

DeBasti, 2023-04-26:
For SMD diodes you can use 1N4148 SOD-123 diodes (I used them in my cheapino build) and for normal through hole diodes you can use regular 1N4148 diodes or the slightly bigger 1N4007 diodes (which are not often used due to the size but they work).
[Example normal diodes](https://www.aliexpress.com/item/1005004491878095.html?spm=a2g0o.productlist.main.3.41117164QXZm4t&algo_pvid=d96d80c6-27cb-45ac-961a-4a9144919149&algo_exp_id=d96d80c6-27cb-45ac-961a-4a9144919149-1&pdp_npi=3%40dis%21EUR%211.8%211.8%21%21%21%21%21%402100b5dc16824920614701785d06d7%2112000029345317729%21sea%21DE%213865699518&curPageLogUid=IKe6Wcj6WbSa)
[Example SMD diodes](https://www.aliexpress.com/item/1005003194674618.html?spm=a2g0o.productlist.main.7.62d9J9FkJ9FkFJ&algo_pvid=ac7c06b8-5d6f-4b66-9a59-5635aeabff4d&algo_exp_id=ac7c06b8-5d6f-4b66-9a59-5635aeabff4d-3&pdp_npi=3%40dis%21EUR%210.81%210.81%21%21%21%21%21%402145274c16824919227917420d06ee%2112000024602962919%21sea%21DE%213865699518&curPageLogUid=NpUBakbnriVG)

tompi, 2023-04-29:
@Throvn I made a build guide with links if you are still interested: https://github.com/tompi/cheapino/blob/master/doc/buildguide_v1.md

Throvn, 2023-05-09:
Yes I very much am. Today some of the first parts arrived. I'm looking forward to assembling it. Thank you for all of the help!