tompi/cheapino #10: Mounting holes

ggppjj, 2023-11-09:
Hello!

Excellent project, I'm very close to having built mine.

I'm thinking of working on a model for a case, and would like to suggest considering mounting holes for the board in the next revision.

tompi, 2023-11-09:
Hey. Sounds good!

Do you have any suggestions for size, number and placement?

ggppjj, 2023-11-09:
I was thinking of four, one on each corner, to allow for a super cheap hacky "throw four screws of different lengths in it for basic tenting" while also providing a good number of mounting holes for more complex projects.

2mm holes should be fine, maybe something like 2.5 for tolerance but I don't know what normal PCB manufacturing best-practice for mounting holes is.

commit f5474241 (2023-12-16): Fixes #10 Added some mounting holes

Also added logo, rounded tracks and tear drops
closed 2024-04-06 by tompi