tompi/cheapino #11: Waiting for drive to deploy...

normausome, 2023-11-10:
Hey guys!

I am having an issue with it being stuck at "Waiting for drive to deploy" when I run `"make cheapino:default:flash"`. not sure why this would be happening. I see the rp2040 being listed when I run "lsusb".

Thanks!

tompi, 2023-11-10:
Hey!

Did you power on the rp2040 while holding down the boot button?

Or hold boot and press reset

tompi, 2023-11-10:
It should show up as a usb drive if you manage to put it in dfu mode

normausome, 2023-11-10:
Hey! so just got it working. flashed it on my mac instead of ubuntu. so i ran `"make cheapino:default:flash"` and it wrote to the MCU,  but now it doesn't type. i see the LED indicator on though!

normausome, 2023-11-11:
would i have to flash it with the RJ45 plugged in?

tompi, 2023-11-11:
No, shouldnt have to do that…

So no keys Are working?

Did you test the miryoku uf2 file?

If neither Are working, maybe post some pics of the board, might be some hw
trouble?

normausome, 2023-11-11:
yes, no keys are working. i need to get a multimeter to test the connection to the keys....

tompi, 2023-11-11:
Send some good resolution pictures of the board. Might be able to spot
problem by visual inspection

normausome, 2023-11-11:
got it! so left side works, but right side doesn't - just had to resolder some of the diodes correctly. sorry this is my first time!

time to figure out what is wrong with the right side.

tompi, 2023-11-12:
Try another rj45 cable and if it doesnt help, post close ups

normausome, 2023-11-13:
ok so what i did was desolder switches and try it out on a new board! thanks for the help!!

closed 2023-11-13 by normausome