tompi/cheapino #16: Encoder locking out keys

ggppjj, 2023-11-29:
Hello!

I've been breaking in the Miryoku layout with the encoder programmed to be arrow keys. I've noticed that using the encoder _sometimes_ causes the backspace key to stop working. Using the encoder again will get it. 

I see in the code that the encoder disables some lines due to a hardware bug, is there some kind of debugging I can help perform to see what part of the code needs adjustment?

ggppjj, 2023-11-29:
FYI, my custom build is accessible from the link in #15. I had the same issue with the Miryoku build on the readme, so I don't think updating sources or my own modifications caused this but I am not running stock.

tompi, 2023-12-01:
I suspect you might have a detenting issue with the encoder? Ive experienced myself that once in a while, the encoder is "stuck" in a position, it doesnt detent properly, leading to strange issues, but turning it in either direction always solves it...

sounds like same problem? Are you a cheapskate like myself and bought a encoder from aliexpress? :)

ggppjj, 2023-12-01:
You know, that's probably it. I used the links for components in the build guide.

I'll see about modifying the lockout code to timeout after a few milliseconds or so and testing that out, I tend to use my encoder a bunch for quick cursor moves and it ends up being really noticable.

tompi, 2024-04-19:
Hey @ggppjj : did you figure this out?
Is it still happening? (and is it fixed by rotating encoder when it does?)

ggppjj, 2024-04-25:
Apologies, I unfortunately wasn't able to power through the learning curve for the keyboard since posting, and the only real time I'm typing with one is at work. I very much intend to get back to using it at some point, but realistically I may now just build up a v2 once I get the time. Thanks for checking in!

It was fixed by rotating the encoder at the time, and I did buy the exact linked parts in the BOM at the time which was the same encoder you had I'd imagine.

tompi, 2024-05-16:
I think this bug will remain in v1... I have not seen it in v2, hopefully it is gone(I made sure to use a separate row for the encoder...)

closed 2024-05-16 by tompi