tompi/cheapino #17: V2 encoder matrix

fauh45, 2023-12-09:
I understand that V2 is still on Beta and not fully stable yet. But I've made ordered it and already made the PCB to test it out. As there's quite a difference in pinout on the encoder from the V1 (and the matrix too on the right side if I'm not mistaken). 

I've copied the qmk matrix, and encoder code and changed some part of it. I've got the matrix to work already but I couldn't get the encoder to work. Tried with the original code, and it doesn't work. I also try to directly check through the matrix by using bitwise and, and still no result. Is there a pinout I'm missing or should it be handled differently?

The code is at (the latest attempt is the one that reads directly from the matrix) https://github.com/fauh45/qmk_firmware/blob/550ae30a64828afa28911ce23fae2954f7aadcc1/keyboards/cheapino/v2/encoder.c#L67

tompi, 2023-12-09:
Hey. Im afk for a few days…

Are you looping through all rows? The old Code skipped the last because of
a hw bug…

tompi, 2023-12-09:
Checked your Code. Try to remove the -1

fauh45, 2023-12-09:
Tried it real quick with the `-1` removed, on both col 9,11 and also 8,10 to no avail. I actually got some kind of output when I tried to map the pins to the keys from info.json, but I don't remember the config that makes it happens.

fauh45, 2023-12-09:
Oh wait, I might've commented the function in `matrix.c` duh...

fauh45, 2023-12-09:
Oh nope... It still not working right now... The idea is that the matrix will "think" of the encoder as just another keys right?

tompi, 2023-12-09:
Did you chek this fw:
https://github.com/tompi/qmk_firmware/tree/cheapinov2/keyboards/cheapino

Seems like your info.json is not correct, not all columns and rows there?

Try the v2 fw and let me know if you get output in qmk console for the
encoder

tompi, 2023-12-09:
oh, and this wip is really wip... try to just build with make
cheapino:default:flash
Started with the v2, but reverted to working directly on cheapino folder...

lør. 9. des. 2023 kl. 10:56 skrev Thomas Haukland ***@***.***

fauh45, 2023-12-09:
Oh shoot, didn't see that v2 build, will try it, thanks!

fauh45, 2023-12-09:
Out of curiosity I tried to super impose your `info.json` onto my code, but the encoder is turning on G and B. I'll try to flash your code directly.

fauh45, 2023-12-09:
@tompi tried your build... yea it doesn`t work, the encoder is spitting G and B, left side is completely off, this is what happened with me by pinning the row and col GPIO number according to the NETS. The one I found to work is,

```ruby
# Got it by matching J_1_LEFT1 pins to J_2_RIGHT1
# e.g. R_COL_1 in pin 2 on J_1_LEFT1 connected to R_ROW_4 in pin 2 on J_2_RIGHT1
r_row1 = 3
r_row2 = 1
r_row3 = 2
r_row4 = 0

row4 = 8
row3 = 29
row2 = 28
row1 = 27

r_col1 = 6
r_col2 = 5
r_col3 = 4

col3 = 26
col2 = 15
col1 = 14
```

(excerp from PRK build for the v2 https://github.com/fauh45/prk-keymap/blob/65b7484ace42417954c12519497f7d550095e98c/cheapino-v2/keymap.rb#L5)

tompi, 2023-12-09:
Just to be clear, i havent mapped keypresses to letters yet…

I had to travel for work, still not home, so this branch should only make
sure you Get data in the matrix, when using qmk console. Still need to map
the matrix to the layout, which is done at the bottom of info.json

fauh45, 2023-12-09:
I see, I'll try to map it first then... Thanks!

fauh45, 2023-12-09:
Got it mapped out, but need to switch some pins out to get it working. Encoder seems to register, but I think the issue of the encoder setting a whole column on is gone already now. Here's some of the console result.

**Random gibberish while rotating**
```
Thomas Haukland:cheapino2:1: r/c 0123456789ABCDEF
Thomas Haukland:cheapino2:1: 00: 0000000000000000
Thomas Haukland:cheapino2:1: 01: 0000000000000000
Thomas Haukland:cheapino2:1: 02: 0000000000000000
Thomas Haukland:cheapino2:1: 03: 0001000000000000
Thomas Haukland:cheapino2:1: 04: 0000000000000000
Thomas Haukland:cheapino2:1: 05: 0000001011010000
Thomas Haukland:cheapino2:1: 06: 0000000000000000
Thomas Haukland:cheapino2:1: 07: 0000000000000000
Thomas Haukland:cheapino2:1: 
Thomas Haukland:cheapino2:1: r/c 0123456789ABCDEF
Thomas Haukland:cheapino2:1: 00: 0000000000000000
Thomas Haukland:cheapino2:1: 01: 0000000000000000
Thomas Haukland:cheapino2:1: 02: 0000000000000000
Thomas Haukland:cheapino2:1: 03: 0001010000000000
Thomas Haukland:cheapino2:1: 04: 0000000000000000
Thomas Haukland:cheapino2:1: 05: 0000001011010000
Thomas Haukland:cheapino2:1: 06: 0000000000000000
Thomas Haukland:cheapino2:1: 07: 0000000000000000
Thomas Haukland:cheapino2:1: 
Thomas Haukland:cheapino2:1: r/c 0123456789ABCDEF
Thomas Haukland:cheapino2:1: 00: 0000000000000000
Thomas Haukland:cheapino2:1: 01: 0000000000000000
Thomas Haukland:cheapino2:1: 02: 0000000000000000
Thomas Haukland:cheapino2:1: 03: 0000010000000000
Thomas Haukland:cheapino2:1: 04: 0000000000000000
Thomas Haukland:cheapino2:1: 05: 0000001011010000
Thomas Haukland:cheapino2:1: 06: 0000000000000000
Thomas Haukland:cheapino2:1: 07: 0000000000000000
```

**One rotation left**
```
Thomas Haukland:cheapino2:1: r/c 0123456789ABCDEF
Thomas Haukland:cheapino2:1: 00: 0000000000000000
Thomas Haukland:cheapino2:1: 01: 0000000000000000
Thomas Haukland:cheapino2:1: 02: 0000000000000000
Thomas Haukland:cheapino2:1: 03: 0001000000000000
Thomas Haukland:cheapino2:1: 04: 0000000000000000
Thomas Haukland:cheapino2:1: 05: 0000000000000000
Thomas Haukland:cheapino2:1: 06: 0000000000000000
Thomas Haukland:cheapino2:1: 07: 0000000000000000
Thomas Haukland:cheapino2:1: 
Thomas Haukland:cheapino2:1: r/c 0123456789ABCDEF
Thomas Haukland:cheapino2:1: 00: 0000000000000000
Thomas Haukland:cheapino2:1: 01: 0000000000000000
Thomas Haukland:cheapino2:1: 02: 0000000000000000
Thomas Haukland:cheapino2:1: 03: 0001010000000000
Thomas Haukland:cheapino2:1: 04: 0000000000000000
Thomas Haukland:cheapino2:1: 05: 0000000000000000
Thomas Haukland:cheapino2:1: 06: 0000000000000000
Thomas Haukland:cheapino2:1: 07: 0000000000000000
Thomas Haukland:cheapino2:1: 
Thomas Haukland:cheapino2:1: r/c 0123456789ABCDEF
Thomas Haukland:cheapino2:1: 00: 0000000000000000
Thomas Haukland:cheapino2:1: 01: 0000000000000000
Thomas Haukland:cheapino2:1: 02: 0000000000000000
Thomas Haukland:cheapino2:1: 03: 0000010000000000
Thomas Haukland:cheapino2:1: 04: 0000000000000000
Thomas Haukland:cheapino2:1: 05: 0000000000000000
Thomas Haukland:cheapino2:1: 06: 0000000000000000
Thomas Haukland:cheapino2:1: 07: 0000000000000000
```

**One rotation right**
```
Thomas Haukland:cheapino2:1: r/c 0123456789ABCDEF
Thomas Haukland:cheapino2:1: 00: 0000000000000000
Thomas Haukland:cheapino2:1: 01: 0000000000000000
Thomas Haukland:cheapino2:1: 02: 0000000000000000
Thomas Haukland:cheapino2:1: 03: 0000010000000000
Thomas Haukland:cheapino2:1: 04: 0000000000000000
Thomas Haukland:cheapino2:1: 05: 0000000000000000
Thomas Haukland:cheapino2:1: 06: 0000000000000000
Thomas Haukland:cheapino2:1: 07: 0000000000000000
Thomas Haukland:cheapino2:1: 
Thomas Haukland:cheapino2:1: r/c 0123456789ABCDEF
Thomas Haukland:cheapino2:1: 00: 0000000000000000
Thomas Haukland:cheapino2:1: 01: 0000000000000000
Thomas Haukland:cheapino2:1: 02: 0000000000000000
Thomas Haukland:cheapino2:1: 03: 0001010000000000
Thomas Haukland:cheapino2:1: 04: 0000000000000000
Thomas Haukland:cheapino2:1: 05: 0000000000000000
Thomas Haukland:cheapino2:1: 06: 0000000000000000
Thomas Haukland:cheapino2:1: 07: 0000000000000000
Thomas Haukland:cheapino2:1: 
Thomas Haukland:cheapino2:1: r/c 0123456789ABCDEF
Thomas Haukland:cheapino2:1: 00: 0000000000000000
Thomas Haukland:cheapino2:1: 01: 0000000000000000
Thomas Haukland:cheapino2:1: 02: 0000000000000000
Thomas Haukland:cheapino2:1: 03: 0001000000000000
Thomas Haukland:cheapino2:1: 04: 0000000000000000
Thomas Haukland:cheapino2:1: 05: 0000000000000000
Thomas Haukland:cheapino2:1: 06: 0000000000000000
Thomas Haukland:cheapino2:1: 07: 0000000000000000
```

So the code might need a bit of change to detect it. I'll push later, and I'll reference it here

fauh45, 2023-12-09:
Here's the PR for the pin mapping and encoder fix https://github.com/tompi/qmk_firmware/pull/3

tompi, 2023-12-16:
Merged PR. Thnx for helping out!

closed 2023-12-16 by tompi