tompi/cheapino #19: m ; p sending incorrect values

lardofpatato, 2024-01-24:
Hi, 
am having a problem with the Cheapino right pinky keys.
The top pinky button is sent middle pinky button in my case O(Colmak) but when using qwerty ;
middle right pinky is sending M 
while M key is sending 

I have tried both keyboards with a single rp2040
I have tried default and custom keymaps
I have tried different ethernet cables 
I have tried different USB cables
visually both my keyboards have all diodes pointing the same direction (black marks to square)

lardofpatato, 2024-01-25:
error occurring in qmk when compiled possible error in matrix

tompi, 2024-01-25:
hey, not sure whats happening here, but need some more details: 

I assume you are compiling from this branch: https://github.com/tompi/qmk_firmware/tree/cheapino

What command do you use to compile?

Sure you didnt change the keymap.json file?
Or the info.json file?

The miryoku is compiled on the basis of this branch, nothing really special, so unless you changed some files, I really dont see how keys could switch codes...

lardofpatato, 2024-01-25:
The default keymap has 2 error. Right ring pinky is set to M and top pinky is set to O. That in conjunction with my having colmak install on windows caused it to freak out more. I got it to work by dragging the uf2 to the rp2040 after using the compile comand.You will need to update the default keymap.Also if you would like I can give you the colmak version of the default keymap I have made.

tompi, 2024-02-18:
Sorry, fixed the default keymap now.

closed 2024-02-18 by tompi