tompi/cheapino #21: more detail on how to work with QMK MSYS so that we get the correct branch for git noobs

jinchoung, 2024-02-18:
hi. so installing qmk msys is very straightforward.

within qmk msys, typing

`qmk setup`

results in a directory qmk_firmware. that's great. and that seems to setup the environment correctly for compiling.

however, i have no idea how get the cheapino branch now.

git clone -b doesn't work neither does git clone --recurse-submodules ....

and not doing qmk setup and doing git clone -b instead results in an environment where i can't compile. so i'm ending up in a situation where either i can compile and can't see cheapino or i can see cheapino but can't compile!

for those of us who are complete noobs to git, can we please get a step by step instruction just to the point where
`qmk compile -kb cheapino -km default`
works?

thanks.

jinchoung, 2024-02-18:
thanks to **honorless** at discord, this procedure works:

`git remote add tompi https://github.com/tompi/qmk_firmware
git fetch tompi cheapinov2
git checkout tompi/cheapinov2`

so the complete procedure would be:

1. instam qmk msys
2. inside qmk msys, type `qmk setup` - this will create a qmk_firmware folder that has cloned the main branch of qmk from git.
3. `git remote add tompi https://github.com/tompi/qmk_firmware`
4. `git fetch tompi cheapinov2`
5. `git checkout tompi/cheapinov2`

and now, you should be able to `qmk compile -kb cheapino -km default` and environment should now be setup correctly for you to make your cheapino firmwares.

**honorless** also says:

"This will leave you in a "detached HEAD" state. Don't panic; you can switch right back to QMK master after you're done building the cheapinov2 firmware by following the instructions `git switch -`."

i have no idea whether i or anyone else will ever need that but better to include it than not.

tompi, 2024-02-18:
Thank you!

Added a fw guide in [8e3e69b14f28139293e2516b3adf0e89525b0df3](https://github.com/tompi/cheapino/commit/8e3e69b14f28139293e2516b3adf0e89525b0df3)

I would really appreciate a photo of your build, especially if it was a v2 :)

closed 2024-02-18 by tompi

jinchoung, 2024-02-19:
hi thomas!

thank you so much for the addendum! that would have really helped me to get to a viable firmware yesterday! i'm sure it's going to be invaluable for others attempting the project in the future.

oh and will totally be happy to send you pics of my proud new build... once i build it! all parts in transit from china lol!

but in perusing the instructions, i just had a feeling that after hours of squinting and soldering, i was going to run into a brick wall when it came to the firmware so i thought i'd do my future self a favor and debug that process first.

oh, sorry but if you get a chance, can you also add a section on how the encoder works? you mention it in the parts list but don't elaborate about it further either for the build or for how things get mapped to it in qmk.

thanks for an amazing and affordable and (looking like) fun project!

jinchoung, 2024-02-19:
oh hey, actually,

where IS the link for cheapino v2? i canceled my parts order and i'll build
that instead.

also, for the keycaps, you say that it's a matter of personal preference
and that's true but can you also include a DEFAULT set of switches that you
used? because i think i'm lucky i was able to cancel my parts order because
i didn't realize that the electrical contact patterns under the switches
are NOT UNIFORM!!! i'm a complete noob to this so details like this are
taking me by surprise and i think it might help others too if there was
just a caveat about this issue.

wow, dodged a bullet there!

thanks.

jin

tompi, 2024-02-20:
Hey. The v2 is not really released yet, but the gerber is here: https://www.reddit.com/r/ErgoMechKeyboards/s/2zsigdfcTC

As for switches, any mx switch should be fine. It really makes no sense to name a default since this is the most discussed topic in the keyboard community:)

For working in an open office space, i like the bluish white silent, at home i like the regular clicky blue switches.

jinchoung, 2024-02-20:
hey thomas,

oh thanks for the info on the switches.

my thought on including a default is a total noob like me can show up and
not realize that the keyboard interfaces are not actually universal. i was
familiar with the idea of tactile vs. linear, etc but had no clue the
actual electrical connection underneath varied. how about just a warning
that they *must* use MX compatible switches and not others? (i don't even
know what the other ones are called) but that's your call of course. maybe
other noobs have more sense than to jump into something like this without
any idea of what they're doing : ) ....  but ok i have selected these:
https://www.aliexpress.us/item/3256806093730983.html?spm=a2g0o.detail.0.0.75de5fc3UFZS3t&mp=1&gatewayAdapt=glo2usa
.

oh cool. thanks for the gerber. actually, i downloaded kicad and exported
the gerber from your pcb project but i'll use your gerber since it's more
likely to have been done properly.

hmm, if the v2 is not released yet, should i wait? is there any
modifications that you're going to make before it drops?

thanks again for your help! really looking forward to putting this
together. i spent a couple of days really getting the keymap right and
successfully compiled my uf2 so it's itching to get uploaded!

jin

tompi, 2024-02-20:
I think the v2 is better. It fixes a lot of minor issues. 

Most importantly it supports 5 pin switches, so you wont have ti clip them, and they will align much easier. 

I wont change anything, except maybe add a logo. 

The only build difference is you need to bridge some solder pads on the right side, it Even says so on the pcb...

jinchoung, 2024-02-20:
oh great! ok. pulling the trigger.

thanks again!

jin

jinchoung, 2024-02-20:
oh wow, thanks for telling me about bridging pads. i would not have known
to do that.

ok, so looking at the board for bridging the pads - you're talking about
the pads labeled, "Solder on right side only" right? sorry, i'm not
clear on what that means.

does that mean only bridge the north/south pads on JP1 (the right-most
north/south pads)? or do i bridge ALL the north/south pads?

thanks.

jin

tompi, 2024-02-20:
The pcb is reversible, so same layout for left and right. 

Right side means the pcb used for the right hand.

jinchoung, 2024-02-20:
oh gotcha. so for right side keyboard bridge all those contacts n/s.

got it.

thanks much.

jin