tompi/cheapino #22: cheapino v2 build! partially working! some problems!

jinchoung, 2024-03-11:
hi!

so i ordered all the parts and the pcb for cheapino v2 and things mostly work!

the index finger columns and the pinky columns on both sides work. and the encoder works!

however, strangely, the middle finger and ring finger columns are completely dead. the weird thing is that they're completely dead on BOTH SIDES!

so that doesn't sound like i wired the switches incorrectly or there are bridges or anything.

any clue what this could be?

also, i've compiled my own QMK setup. but the keys are kind of crazy. the pinky row activates windows menus on the left hand and on the right hand, it brings up emoji stuff. and stuff like home row shift, ctrl, etc don't work on the custom firmware that i make from qmk. any thoughts?

thanks!

will post pics once i get this all sorted!

jinchoung, 2024-03-12:
[image]

oh well, here's the build... though it's still not fully functional, it looks and feels fantastic!

tompi, 2024-03-12:
Hi, the build looks good.

I feel bad about this, because i uploaded an untested gerber(i thought I tested it, but turned out I was just workin on the previous prototype).

Anyway, there is a pretty straighforward fix, you need to run a bodge wire on both sides between the points in the picture:
[image]

I think the best looking solution is to take off some keys and run the wire between switches on top plate, for both sides,
this way it will be hidden, and wont attach to anything. If you prefer to do it on the back(easier) thats fine as well.
I circled 2 pads and 2 holes on one side, just choose the most accessible, they are all connected directly.

This should take care of the missing keys!

Soldering the bottom of the rp2040 is fine. Im not using any of those pins, so its not needed.
Im going to try to make a wireless version with the xiao seed studio, which only has pins on the sides...

For the software, I suspect you have an error in your setup. Could you try starting from scratch, following the firmware guide closely? Its quite new, please report anything strange!

tompi, 2024-03-12:
Oh, and for the wire, anything should work, thinner is probably easier to solder. Twist, flux and tin the ends of the wire before soldering to the pcb will make it easy to work with.

jinchoung, 2024-03-12:
hi thomas,

thank you much for responding.

THIS IS GREAT NEWS! lol. i was convinced i fried something on the rp2040 or did something else that would require me desoldering everything from everything and starting over! so this will take care of both columns that are dead right?

and a bodge wire is an easy peasy fix. WHEW.... really great news.

so i have my project for tomorrow.

and ok, will try to start over on the QMK thing too. everything in the instructions is current and proper for cheapino v2 yes?

thanks again!

jin

tompi, 2024-03-12:
Yeah, will fix both columns. The cheapino uses a special matrix that can double number of keys possible: https://kbd.news/The-Japanese-duplex-matrix-1391.html
So one wire is used for 2 columns(but reversed diode direction for one)

Firmware guide is here: https://github.com/tompi/cheapino/blob/master/doc/firmware.md
I just made one, and specified that you need to substitute "cheapino" with "cheapinov2". So you need to be careful in step 4 and 5

Flashing should be the same as copying uf2 file, so thats not the problem.

jinchoung, 2024-03-13:
woohoo! soldered in the bodge wires on both sides and now everything is running smoooooooth! thanks much for all your help thomas! oh and the qmk programming was user error. i used lshft instead of lshft_t and that was causing the homerow modifiers to fire off constantly.

will post a few more photos when i get settled in.

thanks again.

jinchoung, 2024-03-13:
let's see if this is how you close an issue....

closed 2024-03-13 by jinchoung

jinchoung, 2024-03-13:
[image]
[image]
[image]
[image]
[image]

bodge wire totally fine on the underside!

aliexpress was shockingly reliable but the thing that they messed up on was the rubber feet. so in the meantime, i'm using a box of cards and some non slip shelf liner to try out a tenting setup. still feel pretty weird right now but i'm gettin' there.

all the home row modifiers are working well as well as the layer shifting overlayed on top of the functions on the thumb keys.

this was a fun little project. thanks thomas!

jinchoung, 2024-03-17:
hi thomas,

so the keyboard is working amazingly well and i can't believe how smooth
everything is. but i was wondering if you knew where i can edit the value
to change the steps the encoder jumps per detent.

i looked over the qmk docs and they say to change the ENCODER_RESOLUTION
value in config.h but there seems to be several different config.h files in
the qmk_firmware folder. i made the change in the config.h in the cheapino
folder but that didn't do it. also tried changing the ENCODER_RESOLUTION 4
value in encoder.c but that didn't do it either.

totally non-urgent.

thank you much and hope all is well in your world.

jin

tompi, 2024-03-21:
Hey, great to hear!

Unfortunately, its not possible atm.
I will try to code it up this weekend, shouldnt be much work i think.
(i didnt even know about this value, but I see it defaults to 2, so I
should fix this...)

tompi, 2024-03-23:
> oh wow, thanks for getting back to me.
>
> oh cool, good to hear! please let me know when you get to it. of course,
> whenever you do get to it. no hurry.
>
> thanks again!
>
> jin

Hey, updated my cheapino branch for qmk now.

You probably need to udate to latest qmk(1.1.5) for it to compile...
Just save your own keymap, and do the steps from the firmware guide over
again...

P.S. I tested a little and I kind of think ENCODER_RESOLUTION=1 works best.
Change it in config.h

jinchoung, 2024-03-24:
hi thomas,

oh wow, fantastic! that was fast! thank you so much!

ok, will update qmk_msys and will try a new compile. can't wait to give
this a try.

thanks again and hope you have a great weekend.

jin

jinchoung, 2024-03-24:
hi thomas,

ok, newb question. where is the correct config.h to modify?

is it the one in C:\Users\jinch_000\qmk_firmware\keyboards\cheapino ?

i had a look at that and it doesn't have any entries on the encoder. if
that's the right one, do i just add:

ENCODER_RESOLUTION=1

thanks again.

jin

jinchoung, 2024-03-25:
hi thomas,

okay, so doing some research on my own, i found some things out (maybe) but ran into issues:

so is it correct that i'm editing the config.h file here - C:\Users\jinch_000\qmk_firmware\keyboards\cheapino ?

and is the correct syntax - #define ENCODER_RESOLUTION 1  ?

---------------------------------------

so after that, i tried compiling again with  - qmk flash -kb cheapino -km jin10

but this time, i ran into compile errors. i tried removing my addition with the encoder resolution definition and still came up with errors even before it got to the config.h .... here is what it says:

`arm-none-eabi-gcc.exe (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/cheapino/encoder.c
                [OK]
Compiling: keyboards/cheapino/matrix.c
                [OK]
Compiling: keyboards/cheapino/cheapino.c
                [OK]
Compiling: .build/obj_cheapino_jin10/src/default_keyboard.c
                [OK]
Compiling: quantum/keymap_introspection.c
               In file included from quantum/keymap_introspection.c:5:
./.build/obj_cheapino_jin10/src/keymap.c:17:34: error: macro "LAYOUT_split_3x5_3" requires 36 arguments, but only 1 given
   17 |         [5] = LAYOUT_split_3x5_3(),
      |                                  ^
In file included from ./.build/obj_cheapino_jin10/src/keymap.c:1:
./.build/obj_cheapino_jin10/src/default_keyboard.h:31: note: macro "LAYOUT_split_3x5_3" defined here
   31 | #define LAYOUT_split_3x5_3(k4K, k4J, k4I, k4H, k4G, k0A, k0B, k0C, k0D, k0E, k5K, k5J, k5I, k5H, k5G, k1A, k1B, k1C, k1D, k1E, k6K, k6J, k6I, k6H, k6G, k2A, k2B, k2C, k2D, k2E, k6L, k5L, k4L, k0F, k1F, k2F) { \
      |
./.build/obj_cheapino_jin10/src/keymap.c:17:15: error: 'LAYOUT_split_3x5_3' undeclared here (not in a function)
   17 |         [5] = LAYOUT_split_3x5_3(),
      |               ^~~~~~~~~~~~~~~~~~
 [ERRORS]`

and this is after i updated qmk and redid the steps from your qmk guide.

any insight?

thanks!

jin

tompi, 2024-03-25:
This sounds like you might have an error in your keymap?

Does it compile with -km tompi ?

Send me your keymap if you like, and i can have a look

***@***.***, 90786985

jinchoung, 2024-03-25:
oh wow. yes it did compile. ok, so i changed my keymap since the last time and i guess it didn't like those changes.

huh... the only difference was that i added a "gaming layer" on layer 6 without any of the home row modifiers and i was using TO 6 to jump to layer 6 and then from layer 6, TO 0 to go back to the default layer. that must be the wrong way to do that then... ok. i can play around with that.

------------------

oh, but am i correct about the encoder?

that i'm editing the config.h file here - `C:\Users\jinch_000\qmk_firmware\keyboards\cheapino ?`

and is the correct syntax -  `#define ENCODER_RESOLUTION 1` ?

thank you much!

jin

tompi, 2024-03-25:
Yep, try setting it to 2, and you should need to use 2 «ticks» per
triggered «action»

***@***.***, 90786985

jinchoung, 2024-03-25:
oh interesting... so it's determining how many clicks before a command is
issued? huh.

my issue with the encoder is that every command that is issued jumps too
much... it seems to jump down almost a screen at a time and for me, that's
more disorienting than it is helpful.

so what i wanted to do was, if right now, it's jumping ten lines at a time,
i wanted to try reducing that, maybe even to just one line at a time. more
like how the mouse scroll wheel works.

is there a setting i can adjust for that?

thanks.

jin

tompi, 2024-03-25:
OK, i see, Im guessing you mapped encoder to mouse movement keys like
scroll wheel up/down?

If so, you probably need to adjust these:
https://docs.qmk.fm/#/feature_mouse_keys?id=accelerated-mode

jinchoung, 2024-03-25:
actually, i didn't map the encoder at all. it was just in this state as a
default.

i didn't even know it was programmable. cool. so i've been using the online
qmk configurator for programming the keys. can i do encoder like that as
well?

or if it's a file, what file currently contains the encoder configuration
that i have now?

thanks again.

jin

jinchoung, 2024-03-26:
oh... hang on... is it:

`C:\Users\jinch_000\qmk_firmware\keyboards\cheapino\encoder.c`   ?

according to that, on the home layer, the command a click is set for is KC_PGDN and KC_PGUP ... which would indeed cause the screen to jump in large increments.

ok, will test this out but if this is where the configuration for the encoder lives, i think i can figure out how to make a go of this.

thanks again.

jin

jinchoung, 2024-03-26:
woohoo...

that's the file. i have it all configured now juuuuuuussssssssst how i like it with the encoder doing mouse wheel up and down on the base layer and pgup and pgdown as you had it in layer1 as well as volume controls on layers 5 and 6.

one thing i'd recommend is adding the config.h and encoder.c locations and instructions about encoder functionality in the docs. qmk configurator worked great for me for the keyboard but it doesn't deal with the encoder at all and i had no idea what the default settings even were or how to change it and unfortunately, googling for that precise issue ends up sending you all over the place and would be difficult for a newb to get the required information.

thanks again for the awesome project and helping me to get to this point! as i said before, this keyboard just feels sooooooo luxurious and 'right' and i'm really loving it!

jin

tompi, 2024-03-27:
Sorry for the late reply, but you figured it out yourself i see!

You are right, i will add this to the fw guide. Its not self explanatory.

Happy easter!

***@***.***, 90786985

jinchoung, 2024-03-27:
hey thomas,

no problem. thanks for all your help in getting me to the promised land. :
) nothing but smooth typing ahead!

happy easter!

jin