tompi/cheapino #25: The first two columns on the right board doesn't work

barrosoman, 2024-03-17:
[image]

I'm trying to test the switches before installing the encoder on the right side but the first two columns of switches aren't working, I would like to know if the encoder is necessary for them to work or if I should troubleshoot/change the PCB.

tompi, 2024-03-17:
Should work without encoder. I see at least 2 diodes that Are noe soldered?

tompi, 2024-03-17:
Oh Yeah, and maybe try another cable in case its a crossover cable. I will make a troubkeshooting guide:)

barrosoman, 2024-03-18:
I found the culprit, it was the rj45 socket that I bought.

I didn't realize there was a difference between rj45 sockets and bought any. The pins of my sockets are shifted and they don't connect where they should. 

[image]

As a temporary solution, until the correct sockets arrive, I used two wires to connect the pin to the right hole.

Below are the photos of my solution, on the occasion that someone did the same thing...

[image]

[image]

You may need to cut the path of the pin used in the left board, as it is connected to the first column of switches.

I'm using the Cheapino to write this text and I must say it is gorgeous! Thank you Thomas for the design and greetings from Brazil.

tompi, 2024-03-18:
Cool! However, i would check really careful if this is not a cable issue. Some networking cables have a few wires reversed. 

I never heard of rj45 sockets doing this...

mauroue, 2024-03-20:
> I found the culprit, it was the rj45 socket that I bought.
> 
> I didn't realize there was a difference between rj45 sockets and bought any. The pins of my sockets are shifted and they don't connect where they should.
> 
> ![20240317_211154](https://private-user-images.githubusercontent.com/42954121/313520175-25fe0def-64bf-41ed-9d61-8f3865916cd6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NTE2NjksIm5iZiI6MTcxMDk1MTM2OSwicGF0aCI6Ii80Mjk1NDEyMS8zMTM1MjAxNzUtMjVmZTBkZWYtNjRiZi00MWVkLTlkNjEtOGYzODY1OTE2Y2Q2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDE2MTYwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzNmNmZDlhYjZmZjA3OGY1YjBlMWM4YjVjMTNkM2QxMjYxOGMyODJhOWMwODk5N2Y0YjFlODQ5ZWQzN2FlNDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.WLTo-C11l7hyDztcr2CxHPI7spxYqaIU8Al0_NSPRnw)
> 
> As a temporary solution, until the correct sockets arrive, I used two wires to connect the pin to the right hole.
> 
> Below are the photos of my solution, on the occasion that someone did the same thing...
> 
> ![20240317_211134](https://private-user-images.githubusercontent.com/42954121/313521419-49078ecb-f9f8-4577-ba4e-46df97c54368.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NTE2NjksIm5iZiI6MTcxMDk1MTM2OSwicGF0aCI6Ii80Mjk1NDEyMS8zMTM1MjE0MTktNDkwNzhlY2ItZjlmOC00NTc3LWJhNGUtNDZkZjk3YzU0MzY4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDE2MTYwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY0MzI2MTkwOTJiNWE1Mzc1M2M2YTA1MjczODYzZThlZTllYjcxNmNhMzQ2ZThlN2U3MTUzY2IzOGFiMDBmNjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.9uDvYv9TZOgxW64HqdzWby6N10f09byLH0PmeMbrhIY)
> 
> ![20240317_211125](https://private-user-images.githubusercontent.com/42954121/313521441-1ce2622d-f749-4b33-b18e-5cb60a93d0ca.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NTE2NjksIm5iZiI6MTcxMDk1MTM2OSwicGF0aCI6Ii80Mjk1NDEyMS8zMTM1MjE0NDEtMWNlMjYyMmQtZjc0OS00YjMzLWIxOGUtNWNiNjBhOTNkMGNhLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDE2MTYwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY4MmJmMjAyZTY0NTI1MTNkYWQwODZlNTUzNGUzYmJlNTQ1MTEyMTBkOTIxZmExYzM5OTUxOGMzOGViNjEzOWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Bq5tClla_xRHgqwnnn6YICNH0qmXvFGhnrGBJiOuTBs)
> 
> You may need to cut the path of the pin used in the left board, as it is connected to the first column of switches.
> 
> I'm using the Cheapino to write this text and I must say it is gorgeous! Thank you Thomas for the design and greetings from Brazil.

Hmm. I'm having the exact same issue.
Can you tell me where did you bought the new rj45 sockets? I have no idea what the differences are tbh.

tompi, 2024-03-20:
@mauroue could you please check a few different ethernet cables. I suspect both you and @barrosoman  have got an ethernet crossovr cable: https://en.wikipedia.org/wiki/Ethernet_crossover_cable

mauroue, 2024-03-20:
this is the cable I'm using
[image]

I don't think it is a crossover cable

tompi, 2024-03-20:
Wow, I think you are right. Could you post some closeups of your rj45 sockets?

Also, do you have a multimeter? If so, can you check connectivity from socket to socket with cable plugged in? If you get me a table with how it connects, I can probably create a firmware for your ports

mauroue, 2024-03-20:
[image]
[image]

I tested with a multimeter and it connects directly to the counterpart.
It would be greatly appreciated if its fixable by a firmware update, since I won't need to rely on my terrible solder skills

barrosoman, 2024-03-20:
Yeah, in my case the four back pins are more at the left side and the front pins are more at the right side. 

[image]

At the pictures of the build guide, it seems that the corrected pins are the other way, (back pins to the right, front pins to the left).

[image]

With that said, I think that the problem from @mauroue is different, because his rj45 pins seems correct.

mauroue, 2024-03-20:
I think I soldered it to the wrong holes then haha

barrosoman, 2024-03-20:
I would reapply the solder because atleast at the photo, it looks like some pins are without solder or with just a little of it.

tompi, 2024-03-20:
I agree. But the position seems fine. As long as it Lines up with the white Lines on the mounting side you should be good.

tompi, 2024-03-21:
@mauroue , Rightmost 2 columns are all on the same trace going to the upper right pin on the controller. You should probably check this, and reflow it if it looks funky. 

This trace is also exposed on the square hole directly beneath the rj45 socket(thats why the build guide recommends some tape here if you got metallic housing rj45 socket). You might try to squeeze in a little plastic or something and see if that helps?

mauroue, 2024-03-21:
[image]
I already had a tape underneath.

I checked the connections with the controller pin with a multimeter and they seem fine.
Is there anything else I can try?

tompi, 2024-03-21:
How does the mcu soldering look like? Could you send a pic of both sides?

tompi, 2024-03-21:
Oh, and yeah, to rule out firmware issues, flash the miryoku uf2 file, and test with that.

Try shorting gp0 and gp4 and see what you get(you should get 2 keypresses)

mauroue, 2024-03-21:
Ok, I'll try that later. At work atm.

Btw, thank you for your work and help!

mauroue, 2024-03-21:
Ok, I tried the miryoku file and when I shorted it did nothing. I'm guessing its the mcu then?

tompi, 2024-03-21:
Might be. But could be something on the pcb as well. Did you socket the mcu as suggested in the build guide? If so it would be interesting to pull it out and see if shorting registers then...

If not, maybe send a picture of the mcu?

mauroue, 2024-03-21:
[image]
the other side were pre soldered

mauroue, 2024-03-22:
I swapped to a new MCU and it's working now

thank you for.the help!

tompi, 2024-03-22:
Im glad to hear it!

I think its quite seldom these mcus break, they can really take a beating, but they might break sometimes apparently!

closed 2024-03-22 by tompi