tompi/cheapino #26: Cheapino V2 build. Strange behaviour.

soul-wish, 2024-03-27:
Hi, @tompi.
I've finally soldered everything together but the keyboard works very strange. I can't get what is wrong with it, because this is my first custom build, so I don't have any experience in the process. But I've tried to do everything according to the build guide and in the correct order.
Unfortunately the left half is not working (not a single key). When I connect the right half then some of the keys on the right side are working, but the keycodes are VERY wrong.
I've tried to flash it by using my custom layout, and then tried to flesh it by using default one:

`qmk flash -kb cheapino -km default`

So, the left half is not registering any keypresses.

The right half prints the following (in order of the rows from left to right):
top row: nothing is printed (no keys working)
second row prints: `wesdx`
third row prints: `<tab>q<space>a`
thumbs row: `<nothing>cz`

Here are some photos of my build, maybe you will be able to find something wrong. 
[image]
[image]
[image]
[image]

Thanks for this awesome project, anyway.

soul-wish, 2024-03-28:
It was just a misconfiguration in my local git branch ¯\_(ツ)_/¯

closed 2024-03-28 by soul-wish

tompi, 2024-03-28:
Nice build! Seems you solved your problem before I woke up:) send me a picture when you got it working with keycaps and all!