tompi/cheapino #27: Helper "fork"

MrBarbie, 2024-04-01:
Hi, I stumbled upon this project from some post wanting to do a GB for the PCB as a low cost minimal build.

I was wondering what your helper "fork" looked like to help with building this. I designed some [switch alignment rails](https://www.printables.com/model/582030-switch-alignment-rail) that should work here I think, but unsure if it is overcomplicating it if a simple fork works. 
Scaled down to 3U size should allow the rails to fit perfectly for each column. I did notice your spacing is 19mm instead of the typical 19.05mm. The difference is small enough that the original interrow locks should still work but small differences like that could stack up to be quite the problem over a few columns, so there is a 19mm version too now.

tompi, 2024-04-01:
Hey, my fork was a just a ghetto one, a space bar... not helping very much, but better than nothing...

Yours looks super useful, even though I added 5 pin support in v2, in the spirit of keeping cost low, its very useful to be able to use cheap 3-pin switches from ali, or even harvested from old boards(i have done this myself and its super fun. I would never have figured out i actually prefer clickies if i didnt harvest from a super cheap second hand clicky board... too bad my wife and my collegues does not prefer these switches...)

I hope its okay I link to these from the build guide?

tompi, 2024-04-01:
And yes, I intentionally went a bit smaller, since I felt 19.05 was wasting a bit of space...

closed 2024-04-01 by tompi

tompi, 2024-04-01:
Oh, and do you remember where you saw the thread about group buy? I would really like to recommend them waiting until v2 is released... waiting for last prototype now...

MrBarbie, 2024-04-01:
> I hope its okay I link to these from the build guide?
Sure! I would be glad if it helps! It is why I released the files on printables.

Oooh haha, I thought the spacebar was another method.

Choices is always good imo. Unfortunately I will have to agree with your wife somewhat XD. I find they are mostly too plasticky sounding, but maybe I should try them in a metal board. They definitely are great feedback wise though, but I have moved to loud tactiles.

MrBarbie, 2024-04-01:
> Oh, and do you remember where you saw the thread about group buy? I would really like to recommend them waiting until v2 is released... waiting for last prototype now...

Yeah it is over on [reddit](https://www.reddit.com/r/MechanicalKeyboardsUK/comments/1bqwj06/cheapino_pcb_bulk_order_anyone/), just found the link

commit c80efa0a (2024-04-01): Link to @MrBarbie 3D printed fork in the build guide

Close #27