tompi/cheapino #29: MCU holes are a bit narrow

mrchapp, 2024-04-11:
The holes where the headers for the RP2040 Zero measure 0.762 mm in diameter (according to Kicad). It seems that some square-y pins measure 0.62 mm (per side), so there should be plenty of room for the pins to fit into the holes.

A PCB (v2) recently manufactured by JLCPCB, however, proved difficult to have the MCU headers inserted, probably by extra thick pins and by manufacturing tolerances.

A simple solution here would be to increase the size of the holes for the MCU so that they are wider. For comparison, other footprints seem to use 1.6 - 1.7 mm pads with 1.0 mm holes.

RobertKorteweg, 2024-04-12:
Was planning to order these as well but will wait to see if this prompts a change.

tompi, 2024-04-12:
Hey. I will make them slightly larger for v3, but these have been the same size through v1 and v2, and never had complaints, so I suspect you were unlucky with jlcpcb. 

Anyway: i will keep this issue and change the holes for next revision. 

Also: try to order the rp2040-zero without pin headers, and socket it. This could very well save future problems and avoid the narrow holes.

mrchapp, 2024-04-12:
@RobertKorteweg, it should be fine! Just add a little pressure while inserting the MCU headers.

@tompi, I think the problem was more on the thick pins that came with the RP2040 (Ali item 1005004967926448). Diode legs would have worked perfectly, btw, but I didn't have any sockets at hand.

RobertKorteweg, 2024-04-14:
@mrchapp & @tompi Thanks for the replies, will start to order the components soon.

GameDungeon, 2024-05-02:
Yep, I just had issues attaching the MCU beacuse of the thick pin headers that come with it. Making the holes a little bigger would be quite nice for a V3.

tompi, 2024-05-03:
Checked a bit, and you are right, its a VERY tight fit. Ive been using sockets with thin headers all the time, so havent noticed. Will fix for v3. Consider making them this big: https://www.reddit.com/r/KiCad/comments/1cag0pa/comment/l0rq54a/