tompi/cheapino #32: Vial keymap changes after setting up.

TheMadClover, 2024-04-23:
Built V2, and really enjoyed the process.  However, when using the Vial keymap from releases, I have encounter the oddity that upon leaving the editor (in one case I switched to a different keyboard, in another test, plugged/unplugged).

On layer 1, upon returning to Vial, the keymap was shifted down one row, leaving the top row empty, and layer 2 shifted two rows.

Is this something I've done wrong during build, or something else do you think?

Very new to this, so I've been using the firmware files found here, rather than trying to build my own.

Thanks in advance for any insight.

[image]
[image]

tompi, 2024-04-23:
Whoa, I tested myself and its 100% reproducible.

The crazy thing is the keyboard still uses the same layout, its just
the mapping thats shown wrong in vial...

Probably something I did wrong when I extended layers from 4 to 8...

Ill debug this after work today

tompi, 2024-04-23:
@TheMadClover Hey, I think I got it solved now, could you try the new vial firmware i uploaded(just replace the existing in v2 release), and see if it saves ok now?

tompi, 2024-04-23:
This was the fix, still not sure how it got this effect, but the rowcount in one file was obviously wrong... https://github.com/tompi/vial-qmk/commit/1c8b6644c19f0f4b721ef5afe4001cf808b0c1ff

TheMadClover, 2024-04-23:
Confirmed solved!  Thank you so much, I was starting to think I was going crazy!

P.S.  I've designed a case/switch plate for 3D printing, and would love to share with the community, but I've no idea how to do so...so very new to Git

tompi, 2024-04-24:
A case/plate would be very interesting! If you manage to create a new repo and upload i will link to it. 

I guess you can upload to thingiverse or printables?

Otherwise, if you prefer you can send me the model and i can share it and credit you.

TheMadClover, 2024-04-24:
I'll see if I can figure out the repo.  I've actually got two versions,  a case/plate, and a second version on the printer now with the plate integrated into the case body where the base screws on.  If I can't figure it out I'm a reasonable timeframe, we'll explore the other options.

tompi, 2024-04-24:
I guess the proper git way, if you want to contribute to the cheapino repo, would be that you clone the repo, add your files, commit and push, and then make a pull request.

TheMadClover, 2024-04-24:
I'll try that then!  Thanks again!  Great project!

closed 2024-04-24 by TheMadClover