tompi/cheapino #33: Ghosting when 2 specific keys held and third pressed

torusJKL, 2024-05-05:
I'm testing the Cheapino v2 with the cheapino_vial.uf2  firmware.
I currently have only the left half connected.

When I hold two keys from the same row on column 1 & 2 then pressing a key on row 3 will trigger an additional key on the same row but column 1.

e.g. with QWERTY
Holding down Q & W and then press D will also activate A.
Pressing A will not activate D.

The same ghosting happens when holding two keys from row 3 & 4 and than pressing a key in row 5.
e.g.
Holding C & V and then press T will also activate E.
Pressing E will not activate T.

I have checked the direction of the diodes and they are correct.

Is this an issue with my soldering work, the vial firmware or a hardware bug?

tompi, 2024-05-06:
Unfortunately, this is a hardware bug on the cheapino.

Im working on a firmware fix. The result will possibly affect 2 same-row keys + 2 different column presses all on same hand. This seems very difficult to even achieve physically(you can try yourself pressing 4 keys with 3 fingers), so I hope it will not disturb any existing keymap...
Hopefully ready tonight.

Sorry about this, will get to the bottom of it before v3.

tompi, 2024-05-06:
@torusJKL I made an attempt at fixing this in software, could you try the latest vial fw: https://github.com/tompi/cheapino/releases/download/v2.0/cheapino_vial.uf2
and see if you manage to summon some ghosts with this?

torusJKL, 2024-05-06:
@tompi the original issue is resolved.

The only combination I have found to trigger ghosting so far and that is possible to hit for humans with 5 fingers (although unlikely in day to day typing) is by holding three out of the following four and pressing a fourth of the same group:
E T V Z

With X being the joker if you hold down all four of the above.

torusJKL, 2024-05-06:
I did some testing of the right hand and was able to create two different three key combination that trigger ghosting:
M H ?
M Y ?

tompi, 2024-05-07:
Thank you! Will try fix these, probably tomorrow.

tompi, 2024-05-07:
@torusJKL I could not reproduce the 4 letter combo ghosts, and I assume these are pretty unlikely to be used in a keymap?

For the 4 others I could only reproduc 2, but I think I fixed the other 2 as well, but could need your help in checking... could you retest with latest vial firmware? Thnx again for helping! Oh, and send me a picture, im very curious to see peoples builds!

torusJKL, 2024-05-08:
@tompi I was unable to reproduce neither the three nor 4 key ghosting combos on the right hand side.
Thank you for your quick support!

P.S. picture will follow soon.

closed 2024-05-16 by tompi