tompi/cheapino #35: just a tip - GET A CRIMPER!

jinchoung, 2024-05-12:
just wanted to offer a tip that's made my experience much much smoother - get an ethernet cable crimper!
they just don't make them THIS short and this has really perfected my setup!

[image]

tompi, 2024-05-12:
Thats a good tip! Ill incorporate this into the documentation. 

Ps. What keycaps Are those?

jinchoung, 2024-05-13:
hey thomas,

cool! oh and the key caps are these
https://www.aliexpress.us/item/3256804695049488.html?spm=a2g0o.order_list.order_list_main.93.77261802nU0Wfq&gatewayAdapt=glo2usa
...

i was very nervous pulling the trigger on them cuz i didn't know whether
they would fit the switches but everything fit and works like a champ!

hope all is well with you and yours.

jin

commit cc24907e (2024-07-25): Added some tips/tricks

Fixes #35
closed 2024-07-25 by tompi