tompi/cheapino #36: None of the keys work.

Federino, 2024-05-14:
Hi!

I am having trouble with the cheapini v2 build. I soldered all diodes with the MCU and the ethernet port. After that i decided to solder one switch and test if everything I did works before soldering everything else, but I don't get any output after pressing the switch.

I tryed the default software and the miryoku.uf2 as well, but none of them work.

If i go to vial.rock website it detects the pcb as Cheapino, but when I try the matrix test the button doesn't work.

Do I need to solder all the components before trying?
I am sending pictures as well so you can see the PCB.
[image]
[image]
[image]
[image]

I tryed the troubleshooting points, but I can't seem to make it work.

I also used a tester to check the diodes and they seem fine. I also soldered two switches in order to check if one of them is faulty.

If you need more information please let me know. Thank you!

Federino, 2024-05-15:
Update: It seems like 4 of the keys work. Maybe I have the other diodes not polarised correctly, I will try and resolder everything this afternoon.

[image]

tompi, 2024-05-15:
Hi. You definitely have some bad(a few cold) solder joints which could cause problems.

Try removing excess solder and reflow. When you got it looking nice, let me know if there is still problems and we can debug a specific trace with a multimeter(but I think its solder joints now...)

[image]

Federino, 2024-05-19:
Hi!

I'm really sorry for the late reply, but I didn't have time to fix it until today, and I MADE IT WORK!

It turns out that I had a few things soldered incorrectly:

1) I tryed to use the female conector for the MCU and I did a bad insert of the male "sticks" and they ended up not making contact correctly, which is why none of the keys worked. To solve this, I resoldered the entire chip and It worked perfectly.

2) After I managed to fix that I found out that the top row of the left part was not working either. After some debug of the PCB I found that I did not bridge correctly the right side of the PCB, so I redid that part as well.

After that I managed to make everything work.

I thank you so much for the support and this amazing product!

closed 2024-05-19 by Federino