tompi/cheapino #37: Index and Middle finger columns stop working

BradyJ27, 2024-05-16:
Cheapino v1 btw.

Pretty consistently the two columns on the right board stop working for like 30 seconds or so. I've checked all of the solder points and I think they're all fine. I've also used both QMK and Vial, results are the same.

tompi, 2024-05-16:
Hi. This sounds very annoying. Did you also check the soldering on the left part? Could you send a picture both sides of the left part also? These two columns both end up in one trace going through the rj45 socket ti the other side and into one pin of the mcu.

BradyJ27, 2024-05-16:
Thanks for the quick response! 

I don't see anything wrong with the solder on the left, although it's not pretty I don't think it is causing issues. Attached are two pictures of the left side. 

[image]

[image]

tompi, 2024-05-16:
Looks pretty good. I agree. When this happens, is it related to the encoder? Does it help turning the encoder?

tompi, 2024-05-16:
Ill dig out the schematics tomorrow and find the trace for this column. Typing from my bed and my wife is annoyed;)

BradyJ27, 2024-05-16:
> Looks pretty good. I agree. When this happens, is it related to the encoder? Does it help turning the encoder?

As far as I know, no. It happens when I am just typing and not touching the encoder. I haven't tried to use the encoder to fix it, but I will try when it happens again.

No rush on resolving this, thanks again!

tompi, 2024-05-17:
Hi again, I checked the layout for v1, and these two column are both mapped to column 1(with different diode directions per physical column).

Column 1 is routed through the square rj45 socket holes, so I would recommend you reflow these(especially the one on the right side seems a bit blobby, but hard to tell from pic: might be a cold joint).

The trace ends up at pin 4 on the mcu, counting from top left(marked 29). Try reflowing this as well, maybe add a little solder, seems a tad sparse maybe. 

Also, cant tell if you socketed the mcu, but if you did, it might be worth a try to gently pry it out, and check if leg at 29 is shorter than the others(ie was not pushed entirely in when soldered)

BradyJ27, 2024-05-17:
Thanks so much! I've reflowed the square socket holes and the 29 pin. I don't _think_ the 29 leg is shorter, so I left it for now. I will try this if the issue keeps occurring. So far so good though!

closed 2024-05-21 by BradyJ27