tompi/cheapino #38: Can't compile the qmk Cheapino V2

caioananias, 2024-05-18:
I get this error when i try to compile it, no alteration, only uncommented the lines that make the led change color per layout
```
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s flash KEYBOARD=cheapino KEYMAP=default KEYBOARD_FILESAFE=cheapino TARGET=cheapino_default INTERMEDIATE_OUTPUT=.build/obj_cheapino_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

arm-none-eabi-gcc.exe (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/cheapino/ghosting.c                                                           keyboards/cheapino/ghosting.c:50:9: error: unknown type name 'ushort'; did you mean 'short'?
   50 |         ushort row_num_with_possible_error_cause,
      |         ^~~~~~
      |         short
keyboards/cheapino/ghosting.c:52:9: error: unknown type name 'ushort'; did you mean 'short'?
   52 |         ushort row_num_with_possible_error,
      |         ^~~~~~
      |         short
keyboards/cheapino/ghosting.c: In function 'fix_ghosting_column':
keyboards/cheapino/ghosting.c:69:9: error: implicit declaration of function 'fix_ghosting_instance'; did you mean 'fix_ghosting_column'? [-Werror=implicit-function-declaration]
   69 |         fix_ghosting_instance(matrix, i, possible_error_cause, (i+1)%3, possible_error, error_fix);
      |         ^~~~~~~~~~~~~~~~~~~~~
      |         fix_ghosting_column
cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make: *** [builddefs/common_rules.mk:373: .build/obj_cheapino_default/ghosting.o] Error 1
```

tompi, 2024-05-18:
Hey, entirely my fault.

Im a c# dev by occupation and not that familiar with c/c++. Turns out I inadvertantly used "ushort", which in my dev setup is a macro, but apparently not in yours... I switched to "unsigned short" now, which should be universal.

Try pulling and compiling again?

caioananias, 2024-05-18:
Worked like a charm! Thank you, just a question, is possible to set up per layer led color and have vial support?

tompi, 2024-05-18:
Sure, you just need to get this branch, and uncomment the same code: https://github.com/tompi/vial-qmk/tree/cheapinov2

caioananias, 2024-05-19:
Thanks a lot, now my setup is perfect

closed 2024-05-19 by caioananias

caioananias, 2024-05-19:
[image]
Sorry for the image quality, but here is my build. It's not the prettiest I used some parts I had, but I really like it. It's my first ergo

tompi, 2024-05-19:
Thank you. I think it Looks really good!