tompi/cheapino #40: Stuck keys Issues

ConnorMorehouse, 2024-05-25:
First of all, thank you so much for the work you've put into this project and the continued support! I'm incredibly excited to get this up and working. 

I put together my first attempt this weekend, but I ran into a small problem. On the default key map, the . and h keys show as always pressed like the picture below. When I actually do press them there is no change. 
[image]

I am submitting an issue only because I only see other issues about keys not registering, but not about keys always registering. It has been a long time since I soldered, so I wouldn't be surprised if I did a bad job. I checked with a multimeter and the switches are constantly shorted with no resistance between the leads. I have tried to remove excess solder and to reflow the solder for quite a while, but nothing has changed.

Here's how the solder joints look now. I think I might have done a little bit of damage to the board and switches by trying for too long to fix it, which is why I've stopped. 
[image]
[image]

One thing that seems weird to me is that if I press the stem of the switch towards the top (parallel to keyboard pcb plane in direction of finger extension) of the keyboard, the key no longer registers as pressed and this is consistent. 

I have a feeling that the switches could be broken, but I wanted to check first before I bought new switches or tried to redo anymore work. I don't know what else to try besides new switches 

If there's anymore info you need to help, obviously just let me know. I get the feeling this is probably a quick fix, I'm just at a loss. 

If this isn't the correct place for this request, I apologize. If you could point me in the right direction that would be appreciated.

tompi, 2024-05-26:
Hi! This is a bit rare yes, I have yet to actually hear about broken switches(always on), but this seems like its the case...

Since the Cheapino uses a duplex matrix, a short will typically result in 2 adjacent keys being pressed. And the only place you can realistically acheive a short is on the mcu or on the rj45 connector. 

A few things to try first: all the other keys work on both sides?

Could you test a different cable between halves?

And could you send a pictur of soldering mcu on both sides of the other half, and also the rj45 soldering on the other half?

And i assume these keys turn off when you unplug the right half?
(You can plug/unplug while powered on)

ConnorMorehouse, 2024-05-26:
Thank you for your response! 

Yes, all other keys work as expected. 

With a different cable the same behavior happens. 

Yes, when I unplug the right half the keys turn off.

Here's the picture before I tried reflowing the solder on the mcu
[image]

And here's after 
[image]

I also reflowed the solder on the rj45 jacks, but didn't think to get pictures of the before. Here's how they look now
[image]
Left hand/mcu side

[image]
Right hand/encoder side 

I know I used a little bit too much solder, but I think they should still be viable. 

After reflowing the keys are still always on, they still turn off when I disconnect the ethernet cable. 

It is seeming more and more like it's the switches, which is a bummer. 

Do you think that applying the soldering iron for too long and creating too much could've heated up the components of the switch enough to damage them? Maybe warp the internal metal contacts such that they are now touching when the key isn't pressed? What made me think of this is the fact that when I press laterally on the stem of the switch toward the top of the keyboard, the switch will register as off.

Once again, I want to say thank you for the response and the help!

tompi, 2024-05-26:
Might be. 

Did you try removing the key cap and maybe blow some canned air in it? (Maybe some debris is stuck?)

It can be a bit tricky to desolder a switch, but its not that difficult. 

If you Measure continuity between switch pins, Even unplugged from mcu side, i would try this. 

If you have some solder wick or a solder pump(the ss-02 is very good), Watch some YouTube videos on how to do it. 

What baffles me is that this happened with 2 switches...

How long do you estimate you had the iron on the joint? And what temperature?

ConnorMorehouse, 2024-05-26:
With the right board completely unplugged, there is no resistance between the leads on both of the affected switches.

Compressed air didn't seem to do the trick. 

I have ordered a solder pump and will be trying to remove the switch to test on it's own when it gets here. 

It's hard to remember exactly, but probably between 5 and 10 seconds at 370 C with a chisel head tip. I don't remember specifically spending more time on these switches than any other. There are other switches that I definitely did spend more time on and they seem to be fine. I did the full left board with mcu first and it went slower as I was getting comfortable soldering again. 

I will probably be buying some replacement switches if the switches still have no resistance between the leads off the board. 

Thank you again for your help in this. I recognize this isn't an issue with your work, but probably my fault so the continued help is much appreciated.

ConnorMorehouse, 2024-05-27:
So I have removed the switches from the board and taken the switches apart. I seems that some solder had wicked up the lead and into the switch itself and was shorting contacts in the switch. As the solder was at the bottom of the switch I wasn't really able to remove it, but taking the switch apart and blowing compressed air inside seems to have done the trick. 

I tested all the switches after cleaning and they were working fine. I soldered them back on and tested with the multimeter and they still were fine so I got into vial and tested them and they were no longer stuck on. 

I didn't even think about soldering wicking into switch, nor did I think it was possible but the switches I bought don't seem to be fully sealed around the leads so it's possible. 

Now that I have everything working as expected I am very excited to get set up and used to my new keyboard! I have already ordered another set of switches to build another keyboard with a different feel (linear, this one is tactile). Thanks again for your help, have a great week man!

closed 2024-05-27 by ConnorMorehouse

tompi, 2024-05-27:
Good to hear you figured it out! I would really appreciate a picture of the finished build :)

ConnorMorehouse, 2024-05-27:
[image]
Here is the finished kb, and I typed this on it!

tompi, 2024-05-28:
Nice! The blank keycaps looks good, do you have a link?

ConnorMorehouse, 2024-05-28:
Ya! Here you go
https://www.aliexpress.us/item/3256803835640361.html?spm=a2g0o.order_list.order_list_main.85.77ce1802y5jLK6&gatewayAdapt=glo2usa

glebino, 2024-11-14:
I have same issue, but it's something with mcu. I've unplugged it, and it still shows sticky buttons on matrix test

glebino, 2024-11-14:
Thanks! I think it was it. Accidentally destroyed trying to fix that today 😢. Well, have to order another one...

tompi, 2024-11-14:
If you find the bridge you might be able to fix with some wick. 

Alternatively you might reroute traces with bodge wires, but that requires some hacking;)

glebino, 2024-11-14:
It's stopped booting at all, I think I've overheated the MCU

glebino, 2024-11-14:
Thanks for your help, and amazing job developing all this.