tompi/cheapino #41: Help building miryoku for v1

bunnken, 2024-05-29:
Hi, I've been trying to build miryoku to be able to use QWERTY with it, but so far to no avail. I'm able to build the default cheapino firmware just fine, then I add the `miryoku` branch and merge `tompi/cheapino`. After that I run `qmk compile -kb cheapino -km manna-harbour_miryoku` but I get `Invalid keymap argument.`, then I try `make cheapino:manna-harbour_miryoku` and get

```
QMK Firmware 0.20.6
Making cheapino with keymap default and target manna-harbour_miryoku                           [ERRORS]
make[1]: *** No rule to make target 'manna-harbour_miryoku'.  Stop.
Making cheapino with keymap default_split_3x5_3 and target manna-harbour_miryoku               [ERRORS]
make[1]: *** No rule to make target 'manna-harbour_miryoku'.  Stop.
Making cheapino with keymap lars and target manna-harbour_miryoku                              [ERRORS]
make[1]: *** No rule to make target 'manna-harbour_miryoku'.  Stop.
Making cheapino with keymap tompi and target manna-harbour_miryoku                             [ERRORS]
make[1]: *** No rule to make target 'manna-harbour_miryoku'.  Stop.
Make finished with errors
make: *** [Makefile:415: cheapino:manna-harbour_miryoku] Error 1
```
Other than that I tried forking miryoku_qmk to use github actions and add `tompi/qmk_firmware/cheapino` to `Merge QMK Branches`, but that seems to fail because of merge conflicts. I'm new to qmk so I'm not sure what else to try. P.S. let me know if it's better to ask in the miryoku repo.

bunnken, 2024-06-01:
Hi, after some more research I was able to build miryoku successfully. I found https://github.com/manna-harbour/miryoku/discussions/287 and did the following:

```bash
# first follow the cheapino firmware guide
git remote add miryoku_qmk https://github.com/manna-harbour/miryoku_qmk
git fetch miryoku_qmk
git checkout miryoku_qmk/miryoku users/manna-harbour_miryoku
git checkout miryoku_qmk/miryoku layouts/community/split_3x5_3/manna-harbour_miryoku
```

after that, trying to build miryoku seems to work but I get an error about having to remove IGNORE_MOD_TAP_INTERRUPT since it is now the default. After doing that, the build is successful and everything works as expected.

I'll still wait to see if manna-harbour can confirm that this is the right way to go about it, but for now I'm happy of the result. :)

tompi, 2024-06-05:
Hey, thats great!

Feel free to send a pull request against my qmk fork, otherwise I will replicate your approach and get it in.

P.S. Miryoku is colemak, right? Did you have to edit the keymap manually to change it to qwerty?

bunnken, 2024-06-05:
Error on my part, I meant to say you need to delete `IGNORE_MOD_TAP_INTERRUPT` from `users/manna-harbour_miryoku/config.h` (line 15), so it's not a problem with the cheapino qmk fork.

> P.S. Miryoku is colemak, right? Did you have to edit the keymap manually to change it to qwerty?

There are some options you can set for the build command, for example to get the QWERTY layout you can use `qmk compile -c -kb cheapino -km manna-harbour_miryoku -e MIRYOKU_ALPHAS=QWERTY` https://github.com/manna-harbour/miryoku_qmk/tree/miryoku/users/manna-harbour_miryoku#qmk

tompi, 2024-06-09:
Thank you @bunnken 
Added Miryoku prebuilds to release section, and a miryoku branch in my qmk fork

closed 2024-06-09 by tompi

manna-harbour, 2024-06-17:
The mess with https://github.com/manna-harbour/miryoku/discussions/287 will be sorted out eventually, then you'll be able to build cheapino using the workflow merge feature again. Thanks @bunnken and @tompi for fixing things here for now!