tompi/cheapino #43: Dead link on the ordering guide

eekhof, 2024-06-10:
Hello,

Thanks for the great design. I have been looking at the ordering guide (doc/orderingguide.md), and it seems the link to the silicone bumpers is dead (https://www.aliexpress.com/item/1005002619943801.html) . I am not entirely sure what they are supposed to look like and where exactly they are supposed to go below the PCB, which is why I wanted to ask if it would be possible to update the link to a similar model on AliExpress to make it easier to prevent a missbuy.

All the best

eekhof, 2024-06-10:
My bad, I just saw that the link on the V1 guide is still active (https://www.aliexpress.com/item/1005003044229837.html), maybe it can be added to the buying guide.

tompi, 2024-06-10:
Hi, thats very strange, both links works from my browser...

Is Ali Express filtering items from certain countries maybe? Where are you located?

Anyway, most bumpers should work fine, 5 or 6 mm wide, and 2mm thick are the most common I think.

eekhof, 2024-06-10:
Oh, I am sorry, I should have validated this in a different browser/locale, thanks for the info.

I am located in Germany.

commit 8a4aa232 (2024-07-24): Fix silicone bumper ali link

Fix #43
closed 2024-07-24 by tompi