tompi/cheapino #44: Vial tap-hold problem

mehmetakyurek, 2024-06-11:
Hello, 
When i set home row keys to be shift, control etc. on hold it registers as held while normally typing. I tried to change tapping term but it doesn't change anything, even when i set it to 1000 it was still same. Maybe qmk settings doesn't take effect?

tompi, 2024-06-11:
Ill have a look, dont really use vial much myself...

Could you check if the miryoku firmware(in the release section) works as expected?

Your problem is too short tapping term, right? (your presses holds instead of taps)

mehmetakyurek, 2024-06-11:
I just realized it actually updates tapping term but it's still registered as held while typing fast(if i tap the same key repeatedly it taps, but when i tap different keys it holds), even when tapping term set to 2000.  I tried to change other settings but it's still same. And i also used miryoku and default uf2 files there were no problem.

tompi, 2024-06-11:
I just stumbled across this one, do you think this is related to the problem: https://github.com/vial-kb/vial-gui/issues/180

You are using the quantum keys for shift(kc) right?

Does it work if you toggle "Hold on Other Key Press" ? Does IGNORE_MOD_TAP_INTERRUPT have an effect?

mehmetakyurek, 2024-06-11:
Yes i use quantum keys. IGNORE_MOD_TAP_INTERRUPT actually fixed it somehow. Sorry, i thought i tried everything. Thank you

closed 2024-06-11 by mehmetakyurek

tompi, 2024-06-11:
Great! PS, send an image of your build if you havent! Im collecting them ;)

mehmetakyurek, 2024-06-11:
Sure :D.
[image]
Btw, I'm making a qmk_configurator app for this keyboard in my spare time. I don't know if i will be able to finish it but if you have ideas you can open issues. https://github.com/mehmetakyurek/cheapino-configurator (There's almost nothing yet)

tompi, 2024-06-11:
Thank you! Looking good!

The network cable and encoder knob looks very interesting, is it ali express? do you have links?

mehmetakyurek, 2024-06-11:
Network cable is Ugreen slim, i bought it from Amazon Turkey. This is probably same cable https://www.aliexpress.com/item/1005006505380821.html
And the knob is from CIDOO ABM084 keyboard https://epomaker.com/products/cidoo-abm084

mehmetakyurek, 2024-06-11:
I just wanted to build something in my spare time and I wanted some functions like highlighting configured keys and dragging multiple keys. There were more functions that i forget, i'll probably remember them once i fix my sleep schedule

tompi, 2024-06-11:
Hey, I just noticed you got both v1 and v2 ! Cool

mehmetakyurek, 2024-06-11:
It's so fun to build these ^^. Thank you for this great project. Btw do you think the configurator app would be useful or it's not necessary?

tompi, 2024-06-11:
Im not 100% sure what it does, what do you mean by "highlighting configured keys" and "dragging multiple..."

I think most ppl are happy with the vial interface(I assume you checked out both the web and the local app),
and that if somebody wants to configure more they would compile qmk locally.

But I have not really used the configurator more than tweaking a json keymap, then downloading and using it when compiling locally. Are you using it for compiling also?

mehmetakyurek, 2024-06-11:
Adding borders around the keys that are added to config and selecting multiple keys like selecting icons on desktop. Actually these are not really useful i just thought i can add more feature as i develop. But it's too early yet. I'll let you know if i could make it something worth.

tompi, 2024-06-11:
Wait, you mean indicate which keys changed in the keymap? That would be super useful, and probably for everybody using qmk configurator. But that feature belongs as a pull request in the official configurator repo I think.

tompi, 2024-06-12:
I think it sounds very useful. The qmk configurator could probably need some attention.
If you add features, try to make small iterations, and make many pull requests rather than one large.

E.g. if you manage to indicate which mappings have changed since you loaded a keymap, I would
make a pull request for that here: https://github.com/qmk/qmk_configurator