tompi/cheapino #45: Hard time understanding MCU pin labeling: it is me?

sistrall, 2024-06-12:
Hi!

Cheapino is such a nice project! Congratulations, and thanks a lot for sharing it with the community. :-)

I'm about to build a Cheapino V2, and I'm having some problems figuring out how the MCU pins are used for the right side of the keyboard.

Here is what I tried. I just received all the materials, including the RP2040-Zero. I dropped https://github.com/tompi/cheapino/releases/download/v2.0/cheapino_vial.uf2 on it, and I started tinkering with a pair of tweezers and Vial to understand how the matrix works: I wanted to see which pin couple is used for each switch. When I short pins, I see two switches detected at the same time: I understand that that happens because I don't have the circuit with all the diodes and Cheapino uses a Japanese matrix.

When I short a couple of pins for the left side of the keyboard, everything works as expected. But when I try with the pins for the right side (i.e. pins from 0 to 6) I something different from the schematics that I opened in KiCad. As an example, if I short pin 3 and pin 6 I have the K1 switch on the right side (so the inner one on the top row) and the one beside it detected on the matrix checker: according to the schematics, I was expecting to see those switches detected by shorting pin 0 with pin 3, but that's not the case. Could you help me understand what I'm doing wrong?

tompi, 2024-06-12:
Hi, yeah, two keys, because you are bypassing the diodes in place for this.

No wonder you are having a hard time, I always struggle myself when I revisit this :)

It all stems from the board being reversible, and the rj45 connector reusing SOME of the holes on both sides.
To allow placing the rj45 footprint on both sides, I have to lie to kicad about one of the sides...
(through holes have to have same net to be placed on top of eachother)

I should have fixed the naming at the mcu at least, but I didnt... I made a ticket for it here:
https://github.com/tompi/cheapino/issues/42

Anyway, to test keys, try shoring switch pins.

Oh, and dont forget to send me pictures of the finished build :)

tompi, 2024-06-12:
Basically the order of the rj45 cable wires is reversed when passing between the boards. Fixed in qmk mapping, but is wrong on the schematics(and to a certain degree, it has to be, but I could label it better)

sistrall, 2024-06-13:
Oh, thanks @tompi , now I get it. I think I need more expertise to understand these details just by looking at KiCad schematics! :-)

tompi, 2024-12-29:
This will be handled by #42

closed 2024-12-29 by tompi