tompi/cheapino #46: Vial configuration files not working

seluranen, 2024-06-12:
Hi there! This project is awesome, congratulations!

Just finished building my first one (V2, great guide by the way) and flashed the vial config, and only "cheapino_vial.uf2" worked for me.
Am I doing it wrong? All the other are install3d by boot starting and then drag and drop the file? 

Thank you very much!

tompi, 2024-06-12:
Hi! The other files Are not vial builds, but regular qmk with miryoku or a simple default. 

Are they not working at all?

seluranen, 2024-06-12:
My bad, I was trying to update them with the vial configurator (facepalm)

Works properly!

off topic: any news with the wireless version? <3

closed 2024-06-12 by seluranen

tompi, 2024-06-12:
Good to hear! I havent had time to work on my zmk build... it was mostly just for fun, Im not that big of a ble fan. I will probably take it up again sometime this summer. 

Im almost finished with a really nice case though(for hotswap). 

[image]

Dont forget to send me Photos og your build!

seluranen, 2024-06-12:
Whoa! looks awesome!!!

Keep the good work, and thanks again for everything

seluranen, 2024-06-13:
Not yet totally finished, middle cable still on the way, and my 3D printer died two weeks ago so I'll have to order the plate some place (with hotswap the keys feel really flimsy). But I'm typing this with the cheapino! (at a painfully slow pace...)

[image]

Maybe the thumbs are a little bit high for my liking, I'll try to find a lower keycap (or maybe a keyswitch that still fits in the holes).
And possibly I'll try to move the knob to the thumb area. I'm a noobtard with electronics, so probably I will solder cables form the actual connectors to the new knob... I understand that the position in your design is a product of the symmetry of the board (and My, oh My! What a sexy symmetry. The design is brilliant!)

I'll keep you updated on the progress!

By the way, are you aware of these?

https://github.com/braindefender/KLP-Lame-Keycaps/tree/master

I prefer a more "geometrical" look, but the thumb one may be the solution, an a whole set prints at JCLPCB is like 6€!

tompi, 2024-06-13:
Good job on the build! Looks good, but Yeah, you need some sort of plate for hotswap. 

Those lame keycaps look very interesting! 

Ps. Ive been meaning to fix the searing bright green default on the led. You can map led controls to a layer and adjust color and brightness and it will remember it on future sessions