tompi/cheapino #47: Cheapino not showing up in Vial and Rp2040 not going into DFU mode when using usb-a

Ephraim308, 2024-06-27:
Hi

I'm having a problem with my RP2040. When I use a usb-c port (from my ThinkPad) plugged into my cheapino, I can enter DFU mode, and type. But when I go to Vial (in the app or the website) vial cannot find my keyboard. I've read all the troubleshooting steps, and the firmware guide. However, when I use my usb-a port (on my ThinkPad), I get a green light on my rp2040 but I cannot enter into DFU mode nor can I type, and I still cannot get vial to recognize the cheapino. I made sure all the diodes are the right way, not connections are bridged on the MCU, the only questionable thing is the detached piece of solder or two that is soldered onto the hotswap solder pads, but that is not connected to anything else so I don't think that's the issue. I can send you ample pictures if needed, just let me know. I've tried switching laptops and cables. The only cable I did not switch was the TRS cable, but I don't think that's the problem and I don't have another cable, as I bought the TRS cable from the ordering guide. I've built the V2 non hotswap version and I've bought all the parts from the ordering guide links minus the switches and keycaps.

tompi, 2024-06-27:
Hi! First of all: which uf2 file did you Flash?

Ephraim308, 2024-06-27:
We talked over reddit. For anyone else with this problem, the USB cable I was using did not support data transfer. And vial did not recognize the cheapino because I flashed the prebuilt cheapino uf2 file. Hope this helps!

closed 2024-06-27 by Ephraim308