tompi/cheapino #48: NKRO support?

m0seng, 2024-06-27:
Hi, first of all I want to say this project is awesome. I have built two sets of the Cheapino v2 with a friend and I really appreciate all the details that have been considered in the ordering and build guides.

Apologies if this is not the right repo to ask about this, but would it be possible to add "n-key rollover" support to the Cheapino v2, i.e.: a way to bypass the software 6 key limit? I would like this to be possible as it is important for rhythm games requiring up to 10 fully independent keys (or more!).

I am currently using the Vial firmware, and I can see that the home row and thumb keys are independently and correctly registered in Vial's matrix tester regardless of how many are pressed simultaneously, but when using other keyboard testers, typing etc. a limit of 6 keys is imposed. I have tried the NKRO toggle in the QMK tab of the Vial client and the "NKRO On" keybind but to no avail.

tompi, 2024-06-27:
Hey. I can try to compile with nkro enabled, but how can i test if this works? What keyboard tester Are you referring to?

Oh, and i see qmk docs Are saying you can switch between nkro and 6kro with LShift+RShift+N? https://github.com/qmk/qmk_firmware/blob/master/docs/faq_misc.md#nkro-doesnt-work

Are you sure you tested with the correct modifiers?

m0seng, 2024-06-27:
Hi, thanks for the quick response. Any online keyboard tester such as https://keyboardchecker.com/ works, or even just opening Notepad and trying to hold down many keys consecutively and seeing if each key triggers even with the previous ones held down.

I have just tried the LShift+RShift+N shortcut but it does not seem to work. As for modifiers, I am not entirely sure what you mean, but I am currently testing on a barebones keymap with only one layer of basic letter keys.

tompi, 2024-06-27:
Just verified the same you are seeing, also tried to compile with nkro=true, didnt help.
Stuck with 6kro. Ill ask on the qmk discord tomorrow.

tompi, 2024-06-28:
Think I figured out a way to make it work(not sure why I had to use nkro_force... but it seems to work).
Could you test attached fw and see if it works for you, and i will replace the release version and commit.
[cheapino_vial.uf2.zip](https://github.com/user-attachments/files/16026474/cheapino_vial.uf2.zip)

m0seng, 2024-06-28:
I have just tested it and can confirm it now works once "Enable N-key rollover" is checked in Vial > QMK Settings > Magic.

In my opinion, it might be best left disabled by default (which it is at the moment), since the keyboard itself does not have hardware support for full NKRO, and it can be enabled by users like me who need as close to NKRO as possible.

Awesome stuff, thank you!

tompi, 2024-07-13:
Updated release section with new vial fw now

closed 2024-07-13 by tompi