tompi/cheapino #49: Possible to use OLED screen to show layers?

corbob, 2024-07-04:
I've ordered all the pieces for building my own cheapino, and while I wait, I've been thinking of what I have extra, and what I find helpful about my current ergodox. One such thing is a visual indicator of what layer I'm currently on. I currently have an OLED display that I've used with a Pi Pico 2040. Looking at the suggested MCU, it looks like the pins along the bottom include those that would be required to drive the OLED display. As such, I'm curious if anyone has suggestions for using an OLED display with this setup.

tompi, 2024-07-05:
Hi! There is enough free pins to drive an OLED, the problem is more mechanical, there are no through holes to mount it in...

If you make(3D print?) some sort of holder, that you can attach, maybe screw it in place using the pcb mounting holes, wiring it should not be a problem. This would only work for the left side though, since the right side does not have an MCU...

Another solution is using the RP2040-Zero onboard led to indicate layer. There is code for this in the firmware, but I commented it out since I just found it annoying ;) I know at least one other person using this.

Ill keep this issue and make some through holes for a (o)led for version 3.

tompi, 2024-08-19:
Sorry, I think this is gonna be difficult with the std oleds because of the height of the rj45 socket.

closed 2024-08-19 by tompi