tompi/cheapino #51: [v1] Vial layer amount capped to 7

jarikmarwede, 2024-07-10:
Hello and thank you for designing such an awesome keyboard. I ordered the parts for a V1 months ago when V2 wasn't out, but only recently came around to actually building it. Everything went fine, and I installed the firmware with Miryoku preconfigured and started learning Miryoku.
However, as you probably know, it only includes English letters. I wanted to create two new layers for some German letters in Vial, but I couldn't because the firmware only allows for 7 layers currently (and all of them are used by Miryoku). If I have read this part of the [Vial documentation](https://get.vial.today/docs/firmware-size.html#reducing-dynamic-keymap-layers) correctly, the amount of possible layers is hard coded when compiling the firmware. 
Are there any hardware or software constrains that wouldn't allow the Cheapino to run with e.g. 9 layers?
And somewhat unrelated, would it be possible for you to release the keymap files used to implement Miryoku in the Miryoku firmware for V1, as I am sure I have done something wrong when trying to implement it in Vial?

Thanks again for this awesome project.

tompi, 2024-07-10:
Hi! Yep, vial is originally capped to 4 layers to minimize mem usage on eg old pro-micros with very little memory...

I upped it to 8, thinking this was max for most ppl practically :)

I can up it again to 16 tonight or later this week.

If you wanna try yourself, here is the change from 4 to 8:
https://github.com/tompi/vial-qmk/commit/1bdc0f84f985de23ba2ef09ab3705b88069aea71

Alternatively, if you want to customize the original v1 miryoku build, here are the instructions:
https://github.com/tompi/cheapino/issues/41

jarikmarwede, 2024-07-10:
Nice, that would be really cool. But don't stress about it. When you can find the time, it would be nice to have the limit increased.
And yeah, 0 through 7 makes for 8 layers total, my bad.
Also, thanks for pointing me to the build instructions. I will check them out.

tompi, 2024-07-13:
Hey, I updated the releases sections for both v1 and v2 now, with vial fw that should support 14 layers.
(for some reason 16 gave me a compile error about using to much eeprom mem...)

Check it out, if it works ok please close the issue(I just tested very quickly for v1 and v2)

PS: I guess you need to save your keymap before you flash...

jarikmarwede, 2024-07-13:
Yeah, looks good. Thank you very much.

closed 2024-07-13 by jarikmarwede