tompi/cheapino #52: A few notes for the ordering guide...

corbob, 2024-07-11:
I can absolutely put a PR in to add these notes, but just wanted to get the information out of my head here first:

The RP2040-Zero linked... If you get the one with headers thinking that you can just use them with the sockets... you will be wrong... and if you then think "oh well, I'll just solder it to the board, no big deal", you will be wrong a second time in the same day. The pins on the headers are too big for either the board or the sockets suggested.

The EC11 encoder linked is a listing with about 8 "sizes" or "styles" or however you might refer to it... You will absolutely want to order the actual encoders, and not just the knobs 😂 

Now if you'll excuse me, I need to go desolder some headers 😬

tompi, 2024-07-13:
Hey, I know the headers can be a very tight fit, sorry about this...(in fact there is an open issue for this: https://github.com/tompi/cheapino/issues/29 ), but were you unable to push them in the pcb at all?

The different sizes EC11 encoders are only the height and type of shaft, all will fit the pcb. But yeah, I could specify you need both encoder and knobs...

tompi, 2024-07-13:
I would very much welcome a pull request, otherwise Ill put in an explicit warning about the pin headers.
Feel free to suggest other links as well.

I have used pin headers that came with the rp2040 myself, but only used the two long sides, never tried with all 3...
(I also suspect pcb manufacturer tolerances might play a role, seeing I made the holes a bit too small...)

HennoHimself, 2024-07-15:
I ordered from jlcpcb and the headers are tight but they fit with a good amount of force. I have built two boards now with the same ordering guide and it's fine.

br-lemes, 2024-08-11:
I bought the RP2040-Zero with the pre-soldered header. I needed to cut out the pins at the bottom. But I managed to fit it onto the PCB. Although it was hard and tight, but everything worked well.

[image]

djdanielsson, 2024-09-18:
Off topic a bit, @br-lemes what cables are you using in your picture? I'm looking for something more flexible than the normal cat6 cable that I am currently using.

br-lemes, 2024-09-19:
@djdanielsson the cable is this: https://aliexpress.com/item/1005006184311635.html

commit 56942c9d (2024-11-10): Fixes #52
closed 2024-11-10 by tompi