tompi/cheapino #53: Issues with keys not registering and one key registering 3 different key presses

ConnorMorehouse, 2024-07-21:
Firstly, thank you so much for you help and you work on this project! This is my second cheapino I've made, and I love my first one. 

I am having a strange issue that I can't reason about, maybe two separate issues. All three keys on the ring finger of the left hand do not register. Also, when I press a key in the index column on the left hand, it registers three key presses. It will register the inner index (farthest right), middle finger, and pinky key in the same row as the initial index key that was pressed. It is always in the order of inner index, middle, pinky as well. 

Triple key register example:

   Input            Result                   Input             Result                     Input            Result 
[0 0 0 0 0]      [0 0 0 0 0]           [0 0 0 0 0]      [0 0 0 0 0]            [0 0 x 0 0]      [x 0 x 0 x] 
[0 0 x 0 0]  >  [x 0 x 0 x]    or    [0 0 0 0 0]  >  [0 0 0 0 0]    or    [0 0 0 0 0]  >  [0 0 0 0 0] 
[0 0 0 0 0]      [0 0 0 0 0]           [0 0 x 0 0]      [x 0 x 0 x]             [0 0 0 0 0]      [0 0 0 0 0] 

I have checked all the diodes on the board and they seem to be oriented correctly and registering the correct voltage drops. I have checked the switches and they are working correctly. I have checked that all pads have continuity to the MCU and they do. 

I can't think of what else to do or check. If you could help me understand what might be causing this I'd really appreciate it. 

I will probably be checking over the diodes again tomorrow.

tompi, 2024-07-21:
Hey! Send some high res pictures of both sides of the boards. Might be a short somewhere. 

One thing to check is that there should NOT be continuity between any of the mcu data pins...

Also, which keys do not work at all?(maybe circle them in a pic)

ConnorMorehouse, 2024-07-21:
Thank you so much for the quick response!

Sorry I think the way I worded things was confusing. There is no continuity between the data pins on the mcu, I mean that each switch did have continuity to a data pin, even the switches that never register presses. 

Here is the front of the board. 
[image]

Here's the back of the board. 
[image]

Here's a close up of the back side of the mcu. 
[image]

And here's the column of keys that never register a press. 
[image]

I wouldn't be surprised if it was a short that I'm just not seeing. 

Let me know if there's anything else I can provide to help, and thanks again for your help.

tompi, 2024-07-21:
Hey. Im travelling so a bit limited;) but Linda Looks like there is a short on the underside of the mcu, between the unused encoder Hole and one of the pibs(can only see one side...)?

I think this unused Hole is gnd, so it might render one column permanently on/off.

ConnorMorehouse, 2024-07-21:
No worries. I'll try to clean that up soon and get back to you with the results. 

Thanks again, and have fun on your travels!

ConnorMorehouse, 2024-07-21:
I have reflowed the solder on the suspect pin, but the behavior has not changed. I will be downloading the kicad files to see if If can better understand what connections are being made and what the intended connections are. 

I also retested all of the diodes and they seem to be oriented correctly and have the correct voltage drops so I think they're fine. 

I very much appreciate your help, but this is not an urgent issue. I have another working cheapino and don't even have the key caps in yet for this new one.

tompi, 2024-07-22:
This looks like it would pull down row2:
[image]

Is problem nr 1 still there after this bridge is removed?

ConnorMorehouse, 2024-07-22:
Here's an updated picture of the mcu pins. I think I have fixed that issue, but the behavior of the keyboard has not changed. 
[image]

tompi, 2024-07-25:
Hey, do you still experience both issues? (one column not working, and middle key triggering 3 key-presses?)

Since the soldering looks ok, I was just wondering, did you test the keys with a tweezer/wire, without switches?

ConnorMorehouse, 2024-07-26:
Yes, unfortunately the behavior hasn't changed at all. I tried with a small wire over all the keys and got the same results. The column 2 pin has continuity to ground. 

I am looking over the schematics in kicad and I don't see any components that are supposed to be grounded other than the RJ45 shielding. And the only components that have continuity to ground are connected to pin 2. Unless I damaged the solder mask very badly somewhere but can't see it after looking very closely for a couple hours, then the only way I can see the pin connecting to ground is through the mcu. 

I have also started getting errors from my computer that the usb port it's connected to is drawing too much power and turning of the usb connector. It's not super consistent, but almost 50% of the time probably.

tompi, 2024-07-26:
OK, that should be pretty easy to verify: just check for continuity between gnd and the pin with the mcu out of its slot.

If there is still continuity, the problem is on the pcb, else its on the mcu.

ConnorMorehouse, 2024-07-31:
I apologize for the delay, as I have been busy. I removed the mcu today and there is continuity between the pin and ground pin with the mcu removed. I have ordered another. 

Thank you so much again for your help in diagnosing this. I've really enjoyed my first and I look forward to enjoying my second as well. I appreciate your continued support and development on the project, and am looking forward to v3 :)

closed 2024-07-31 by ConnorMorehouse