tompi/cheapino #54: QMK keyboard.json not found

paulomp90, 2024-07-23:
Hello, first of all thank you for your work on this project 👏 

I have a problem with `qmk`, I just checked out `qmk` repo. Then I followed the instructions, in the firmware guide (because I have a mac, I started on step 3) [here](https://github.com/tompi/cheapino/blob/master/doc/firmware.md#setting-up-and-compiling-qmk)

Unfortunately I am always getting the error `keyboard.json` not found.

Does anyone know what I am doing wrong?
Thanks in advance.

[image]

tompi, 2024-07-23:
Hi. Thats just a warning. The actual error is you Are missing a compiler. 

I think you should run qmk doctor and see what it says

paulomp90, 2024-07-23:
@tompi thank you 👏  
I did not knew about the `doctor` command, some packages we're missing because of the checkout 😅

Now I need to finish soldering 😄  I will close the issue

closed 2024-07-23 by paulomp90

tompi, 2024-07-24:
Cool. Dont forget to send me a picture of the finished build! :)