tompi/cheapino #59: Cheapino With USB cable

azzamsa, 2024-08-19:
Hi 👋🏽,

I love the ergonomic design of the Cheapino (1 MCU) and its thumb cluster. However, the RJ45 connector is a bit too large for my taste. Would it be possible for me to fork and modify it to use a USB cable instead? How difficult would that be, considering I have no previous experience with KiCad?

Thank you for creating the Cheapino. I've always dreamed of making split keyboards more accessible to communities, especially in countries where the cost can be a significant barrier. When I saw the announcement of the Cheapino, I was thrilled. 🥞

tompi, 2024-08-19:
Hey, thank you.

There are at least 2 serious issues with using a usb c cable:

1. Most cables only have 4 strands: pwr/gnd and data send/receive. You would have to purchase one of the expensive ones to use for your keyboard... Pppl would also be very annoyed about this....
2. It might not be obvious which port is for computer and which is for other half
3. The 12 strand reversible usb-c connector is a nightmare to solder(its only available as surface mount I think, and you need a reflow table/oven or heat gun with fluid solder right?)

where 1 is a dealbreaker for me...
Ref 3, you might be able to find a breakout board, but there is no standard as far as i know...

If you wanna give it a go, find another design that uses usb-c cross halves(ive seen them on reddit), and copy the footprint from that into the cheapino design!

closed 2024-08-19 by tompi

azzamsa, 2024-08-19:
The soldering part sounds more painful than I expected.

After inspecting the community builds, turns out there is a thin variant of the cable, I have checked and it is widely available locally. So I can use this cable instead. It has the same diameter just like the TRRS cable.

For the bulky port, I can just hide that inside a 3dp case.
With the same height, I think I can fit that bulky RJ45 port inside.

Btw, what do you have in mind other than RJ45 during Cheapino development, and what are their cons?

---

[image]

[image]

[image]

tompi, 2024-08-24:
Sorry about the late reply, but rj45 was really key to the whole concept, since not only is it cheap and widely available, most ppl already have many of these cables laying around...

I think a lot of ppl also use rj11(old telephone cable). These are more flexible, but only have 4 pins, so only really suitable for designs with mcu on both sides...