tompi/cheapino #60: Encoder button always on

ihales, 2024-08-25:
Thanks for sharing such a cool project! I built my first last night and loved it.

I've ended up with one problem- appears that the encoder button is always depressed, even when the right and left halves aren't even connected.

I don't see any obvious problems, and I'm not sure the best way to troubleshoot. I'd love any pointers.

Here are photos of the front and back of the left board.

[image]

[image]

[image]

tompi, 2024-08-25:
Hi! Dont know which firmware you are using, but I would suggest trying the vial fw, without right half connected.
If you go to the test section there, is the encoder indicating press?

The pins on the mcu seems like they are not solidly connected to the hole? Maybee too little heat, or no flux? But that should not cause the encoder issue anyway...

P.S. Im a bit baffled by the bumper placement, are yours lower than your hotswap socket height?

ihales, 2024-08-25:
Thanks so much for the response. Using the vial fw with only the left side connected, this is what I get:
[image]

Thanks for the feedback on the MCU soldering - this is my first real soldering project so the suggestions are super helpful! While working on it, I did accidentally bridge one of the pins to the big encoder hole (on the left, if you're looking at the bottom), but I think I got that cleaned up.

And yeah, with the bumper placement, I put a first round directly on the PCB, but then there were a couple of places where it wasn't high enough, so I tried putting bumpers on the hot-swap sockets. I imagine it'll probably stress the sockets, so probably not ideal. I might try to print a case to help out.

Anyway, if nothing comes to mind, no worries. I have parts for another and I'll just give it another shot!

tompi, 2024-08-25:
Ok, seems like you socketed the mcu?
Worth a try to take it out carefully, and then plug in usbc cable in the rp2040-zero, and see if you get the the same result in vial.

If you do, there is something wrong on the rp2040, either your soldering, or the mcu itself. You can then try with your other mcu, should work...

If it doesnt happen, should be something on the pcb.

If you have a multimeter, you could check that there is NOT continuity between any pins on the mcu socket...

ihales, 2024-08-26:
I tried taking the MCU out and several pins came undone (the sketchy soldering you mentioned, so doubt). The MCU alone was fine and the sockets checked out. I just put it back together and resoldered the MCU and it started working!

Thanks for the suggestions and help. Here's a picture of the finished product!

[image]

closed 2024-08-26 by ihales

tompi, 2024-08-26:
Great! 

Looks really good with those keycaps. 

Good luck with the typing, now the hard part begins;)