tompi/cheapino #61: question about low profile

djdanielsson, 2024-08-26:
you say in your buy list not to buy the low profile sockets, but i was interested in trying to go low profile switches, could you explain why that will not work here or if there are some options to go low profile?

tompi, 2024-08-27:
Do you mean choc-switches? They have another pin-configuration, so the pins will not physically fit the pcb...

There are some low profile MX-switches, and if they are compatible with standard mx, they should fit. 

Im also interested in this, since it seems smart to get the keyboard as low as possible if you are resting your hands on your desk while typing, which I am for the most part...

Any switch in particular you are looking at?

Keycaps also plays a role, I have been trying to order some low profile like these:
https://www.aliexpress.com/item/1005004840360158.html
but without legends, but they never arrived... guess it was a fake listing...

djdanielsson, 2024-08-27:
I see what you mean, now that I look back at the low profile switches I had seen do seem to be all choc-switches. I will look around some more but if I cannot find one I will just give standard ones a try. If I find something I will post back here, if not I will close out the issue.

djdanielsson, 2024-08-27:
I spent a bit of time looking and wasn't able to find any, so I ordered standard size but I wanted clicky so didn't go with the ones you suggested. I guess we will see how it goes in a few weeks once it all comes in.

closed 2024-08-27 by djdanielsson

SadWitchBoards, 2024-09-01:
Unfortunately, the pins for any low profile sockets are not compatible with MX hotswaps. At least, to my knowledge. Choc V1 switches are the most widely used but choc V2 are becoming more popular in the traditional keyboard space due to their better sound/feel and use of materials such as POM. I'd really like to see more ergo boards compatible with Choc V2 since it does provide the MX stem, making keycap selection far wider. The only change I've ever thought I'd life on the Cheapino was a choc version whether it be V1 or V2. Gateron low profit switches are okay but Kailh really is ahead of the pack here with variety and feel.