tompi/cheapino #62: not a git repository

CannonFodderSE, 2024-08-28:
Greetings,
Trying to  follow the instruction at tompi/cheapino [Setting up and compiling qmk](https://github.com/tompi/cheapino/blob/master/doc/firmware.md#setting-up-and-compiling-qmk) and I get the error below.  Looks like the source is gone.

`$ git remote add tompi https://github.com/tompi/qmk_firmware
fatal: not a git repository (or any of the parent directories): .git
`

Any suggestions?

Thanks,
Randy

tompi, 2024-08-29:
Hey, I think you either skipped step 1-2 or failed to go into the qmk_firmware folder that should be there after step 2?

tompi, 2024-09-14:
Hey @CannonFodderSE , did you figure out what was missing?

CannonFodderSE, 2024-09-15:
Yes, I was running it from the wrong directory.

closed 2024-09-15 by CannonFodderSE