tompi/cheapino #63: QMK generates weird layout

os-mey, 2024-09-04:
First of all, thank you so much for all your work!

I believe I assembled the keyboard correctly as it works perfectly with Vial. But I can't get any layout to work with qmk.

I followed the [firmware guide](https://github.com/tompi/cheapino/blob/master/doc/firmware.md) on a mac. I installed qmk using `python3 -m pip install qmk` and some of it's dependencies manually.  
Running `qmk flash -kb cheapino -km default` seems to be working:
<details>
<summary>Output</summary>

        Ψ Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s flash KEYBOARD=cheapino KEYMAP=default KEYBOARD_FILESAFE=cheapino TARGET=cheapino_default INTERMEDIATE_OUTPUT=.build/obj_cheapino_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

        ⚠ cheapino: Build marker "keyboard.json" not found.
        Generating: .build/obj_cheapino_default/src/info_deps.d                                             [OK]
        ⚠ cheapino: Build marker "keyboard.json" not found.
        arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
        Copyright (C) 2020 Free Software Foundation, Inc.
        This is free software; see the source for copying conditions.  There is NO
        warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
        
        Generating: .build/obj_cheapino_default/src/config.h                                                [OK]
        Generating: .build/obj_cheapino_default/src/keymap.c                                                [OK]
        Generating: .build/obj_cheapino_default/src/info_config.h                                           [WARNINGS]
         |
         | ⚠ cheapino: Build marker "keyboard.json" not found.
         |
        Generating: .build/obj_cheapino_default/src/default_keyboard.c                                      [WARNINGS]
         |
         | ⚠ cheapino: Build marker "keyboard.json" not found.
         |
        Generating: .build/obj_cheapino_default/src/default_keyboard.h                                      [WARNINGS]
         |
         | ⚠ cheapino: Build marker "keyboard.json" not found.
         |
        Compiling: keyboards/cheapino/encoder.c                                                             [OK]
        Compiling: keyboards/cheapino/ghosting.c                                                            [OK]
        Compiling: keyboards/cheapino/matrix.c                                                              [OK]
        Compiling: keyboards/cheapino/cheapino.c                                                            [OK]
        Compiling: .build/obj_cheapino_default/src/default_keyboard.c                                       [OK]
        Compiling: quantum/keymap_introspection.c                                                           [OK]
        Compiling: quantum/quantum.c                                                                        [OK]
        Compiling: quantum/bitwise.c                                                                        [OK]
        Compiling: quantum/led.c                                                                            [OK]
        Compiling: quantum/action.c                                                                         [OK]
        Compiling: quantum/action_layer.c                                                                   [OK]
        Compiling: quantum/action_tapping.c                                                                 [OK]
        Compiling: quantum/action_util.c                                                                    [OK]
        Compiling: quantum/eeconfig.c                                                                       [OK]
        Compiling: quantum/keyboard.c                                                                       [OK]
        Compiling: quantum/keymap_common.c                                                                  [OK]
        Compiling: quantum/keycode_config.c                                                                 [OK]
        Compiling: quantum/sync_timer.c                                                                     [OK]
        Compiling: quantum/logging/debug.c                                                                  [OK]
        Compiling: quantum/logging/sendchar.c                                                               [OK]
        Compiling: quantum/logging/print.c                                                                  [OK]
        Compiling: quantum/matrix_common.c                                                                  [OK]
        Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
        Compiling: quantum/main.c                                                                           [OK]
        Compiling: lib/printf/src/printf/printf.c                                                           [OK]
        Compiling: drivers/eeprom/eeprom_driver.c                                                           [OK]
        Compiling: drivers/eeprom/eeprom_wear_leveling.c                                                    [OK]
        Compiling: quantum/wear_leveling/wear_leveling.c                                                    [OK]
        Compiling: platforms/chibios/drivers/wear_leveling/wear_leveling_rp2040_flash.c                     [OK]
        Compiling: quantum/color.c                                                                          [OK]
        Compiling: quantum/rgblight/rgblight.c                                                              [OK]
        Compiling: quantum/rgblight/rgblight_drivers.c                                                      [OK]
        Compiling: quantum/process_keycode/process_rgb.c                                                    [OK]
        Compiling: quantum/led_tables.c                                                                     [OK]
        Compiling: lib/fnv/qmk_fnv_type_validation.c                                                        [OK]
        Compiling: lib/fnv/hash_32a.c                                                                       [OK]
        Compiling: lib/fnv/hash_64a.c                                                                       [OK]
[... 153 lines cut ...]
|---|---|---|---|---|                   |---|---|---|---|---|
|   |   |   |   |   |                   |MO1| q |SPC| a |GUI|         (SPC --> space, GUI --> cmd key/KC_LGUI)
|---|---|---|---|---|---|---|   |---|---|---|---|---|---|---|
                |   |   |   |   |   | c | y |
                |---|---|---|   |---|---|---|
```
Turning encoder to the right: "fv"
Turning encoder to the left: "vf"
Pressing encoder: "r"

Layer1
```
|---|---|---|---|---|                   |---|---|---|---|---|
|   |   |   |   |   |                   |   |   |   |   |   |
|---|---|---|---|---|                   |---|---|---|---|---|
|   |   |   |   |   |                   | 2 | 3 |   |   |   |
|---|---|---|---|---|                   |---|---|---|---|---|
|   |   |   |   |   |                   |   | 1 |SPC|   |   |
|---|---|---|---|---|---|---|   |---|---|---|---|---|---|---|
                |   |   |   |   |   |   |   |
                |---|---|---|   |---|---|---|
```
Pressing encoder: "4"

I get similarly unexpected results with the other provided examples "lars" and "tompi" in "cheapino/keymaps/".

tompi, 2024-09-04:
Hey! Are you sure picked the v2 branch? I suspect you Are compiling the v1
version…

***@***.***, 90786985

os-mey, 2024-09-04:
Works like a charm, thank you so much!

closed 2024-09-04 by os-mey

tompi, 2024-09-05:
Nice! Dont forget to send me a picture of the finished build ;)

os-mey, 2024-09-05:
[image]
Here is my built :)

tompi, 2024-09-06:
Thank you. Looking great!