tompi/cheapino #64: Question: Are the magnetic USB-C safe?

Batres3, 2024-09-06:
I've read that magnetic USB-C is not compliant, and I've heard of a couple of people who have had issues. Has anyone here had any issues? If so are there any alternatives?

soul-wish, 2024-09-06:
I have two Cheapino V2s with magnetic connectors, and so far, I haven’t had any issues with them. So I'm not sure what you're referring to @Batres3

Batres3, 2024-09-06:
@soul-wish sorry I wasn't very clear I updated the message

tompi, 2024-09-07:
Did not have any issues... Was a guy at reddit which I suspected might have some bad magnetic usb-c cables(his board was heating up...), but he never replied back...)

Ive been using mine(branded "Elough") for years without issues, I love them!

Only time there was a problem was when a diode leg got stuck shorting the pads. Did not cause any damaged, just removed it...

Batres3, 2024-09-07:
Ok! I think it should generally be ok to use magnetic cables with keyboards because of the extremely low power draw, still wanted to bring this up just so people know.

Also, thanks for adding the warning @tompi !

closed 2024-09-14 by tompi