tompi/cheapino #65: custom miryoku qwerty configuration

bpingris, 2024-09-08:
Hi! i've built the keyboard and used the miryoku qwerty configuration from the [release](https://github.com/tompi/cheapino/releases/tag/v2.0), works well!
however I had some issues with some layers with the navigation or mouse not following the vim bindings (jkl' and not hjkl), but by looking in miryoku_babel repository i found that there are some [VI alternatives layers](https://github.com/manna-harbour/miryoku_babel/blob/master/tangled/qmk/miryoku_layer_alternatives.h#L271) that seems to be what i am looking for

how can I build a new configuration with this alternatives set up for the cheapino? I try to find some information online but I struggle to [understand it](https://github.com/manna-harbour/miryoku/discussions/85), furthermore how can I customize the encoder behaviour?

thanks a lot, really happy with the keyboard so far!

tompi, 2024-09-08:
Hey!

You have two options:

1. checkout the miryoku branch from my qmk repo and do your changes there and compile/flash
2. use vial and map everything manually. 

1. is probably best. Let me know if you Get stuck somewhere

bpingris, 2024-09-08:
are you referring to the [zmk config repo](https://github.com/tompi/cheapinov2-zmk-config)? I dont see one for qmk on your profile, nor a miryoku branch on the zmk repo (or the cheapino one either)
about vial it does not detect the keyboard with the miryoku layout, but it did when I had installed the basic [cheapino_default.uf2](https://github.com/tompi/cheapino/releases/download/v2.0/cheapino_default.uf2)

bpingris, 2024-09-08:
for some reasons I couldnt see your [qmk repo](https://github.com/tompi/qmk_firmware), I found it after reading your [firmware.md](https://github.com/tompi/cheapino/blob/master/doc/firmware.md#build-your-own-miryoku) file, will check on it now!
thanks!

bpingris, 2024-09-08:
if it can help here's how I compiled my layout:
```sh
git clone git@github.com:tompi/qmk_firmware.git
git switch cheapinov2-miryoku                                                                                                                                                                                  │
./util/docker_build.sh git-submodule
# i've edited the util/docker_build.sh to add my environment variables to customize the miryoku_layout, 
./util/docker_build.sh cheapino:manna-harbour_miryoku
# copy/paste the uf2 file created into the cheapino
```

```sh
"$RUNTIME" run --rm -it $usb_args \
	$uid_arg \
	-w /qmk_firmware \
	-v "$dir":/qmk_firmware \
	-e ALT_GET_KEYBOARDS=true \
	-e SKIP_GIT="$SKIP_GIT" \
	-e MAKEFLAGS="$MAKEFLAGS" \
    -e MIRYOKU_ALPHAS=QWERTY \ # the custom miryoku layout
    -e MIRYOKU_NAV=VI \
    -e MIRYOKU_MOUSE=VI \
    -e MIRYOKU_MEDIA=VI \
	ghcr.io/qmk/qmk_cli \
	make "$keyboard${keymap:+:$keymap}${target:+:$target}"
```

closed 2024-09-08 by bpingris

tompi, 2024-09-12:
Hey, I never knew about that docker option. Seems very useful as its a hassle to setup local dev...
Definitely gonna check that out!