tompi/cheapino #66: Possible to handwire?

ashok-arora, 2024-09-12:
Would to be possible to handwire the keeb, including the rj45 connector and the mcu connector?

ashok-arora, 2024-09-13:
Oh okay, makes sense, thank you. 😄

closed 2024-09-13 by ashok-arora

azzamsa, 2024-09-26:
> the rj45 connector needs to be attached to something, and there is also very little wiggle room in the case for loose wires.

@ashok-arora  Unless of course you take the Cheapino layout and make your own 3d case

tompi, 2024-09-26:
@azzamsa  I would argue that at that point it is no longer a cheapino ;)

azzamsa, 2024-10-12:
@ashok-arora there are many big Cheapino case that is suitable for hand-wire project.
Some thing like [this](https://www.printables.com/model/879047-two-case-designs-for-cheapinov2-split-ergo-keyboar). You can also type "cheapino" in the search bar, as there are many other cases.

[image]