tompi/cheapino #69: Im new to this and this was my first ever build, so I have a couple questions

jcs021102, 2024-09-16:
1) 
[image]
These are my functioning keys (light) vs non functioning (dark)

MCU Pics:
https://imgur.com/a/MEPylSY

2) for miryoku_zmk, what's the board name? I want to try Miryoku w/Colemak mod Dh, but I have no clue what the settings are supposed to be

tompi, 2024-09-17:
Hi!

For nr 1, could you post some pictures of both sides of both pcbs?

Especially the bridge pads on the right side. Wondering if its a combination of missing bridges and  wrongly oriented diodes. 

For nr 2, cheapino is not compatible with zmk, but you can use miryoku on it. Did you check out the firmware guide?

jcs021102, 2024-09-17:
[image]
Updated matrix test (As I had forgotten to solder a keycap)

Images with keycaps attached
https://imgur.com/a/t9qJYaK

Ill send an image without the keycaps in a minute

jcs021102, 2024-09-17:
yeh, diodes are all well and good, I also tested the rj47, as well as the microcontroller. I dont have another rj47 on hand, but its the one I use to connect my landline, so it should be functional.

Im unsure what it means to short the bridgespots

tompi, 2024-09-17:
I mean shorting the soldering pads referred to in section 3 of the build guide. 

Are you able to trigger some of the missing keys if you short these?(just do 2 and 2 at a time)

Rj45 cable could still function as Ethernet but fail in the cheapino if its a crossover...

jcs021102, 2024-09-18:
I found that when shorting between the mcu and the hotswap soldering pads of the functioning keys to be connectivity  present, and of course the non functioning ones do not. Everything else seems to work, just fine.

It is worth nothing that I did solder on 6 hotswap pads in the places where the keys connected, but removed them due to particular reasons. Could this affect things?

tompi, 2024-09-19:
Hey. I meant the 7 solder pads on the right side. Check section 3 of the build guide. 

If you short 2 of these at a time, Are you able to trigger missing keypresses?

jcs021102, 2024-09-19:
I'll check when I get home tonight. Just to make sure I understand, if I
short the 7 solder pads keypresses should occur, and thus if they dont, it
would imply an issue with the board itself correct?

tompi, 2024-09-19:
If shorting triggers it means the fault is on the right side, probably diodes put the wrong way. 

If it doesnt short, you should start searching on left side: short the pins of the rj45 connector. If it shorts its cable or right connector. 

If it does not short its pcb or mcu

jcs021102, 2024-09-19:
shorting the 7pads led to these firing:
[image]

the other far right keys still work when manually pressing them

tompi, 2024-09-20:
I should have been more clear. I meant to short every 2 combinations of these pads. So start with the first pad, short it with the other six. Then take the next and short it with the following 5 and so on. So 6+5+4+3+2+1 combinations.

jcs021102, 2024-09-20:
Yes, that is exactly what I did

tompi, 2024-09-20:
ok, it should not be possible that some keypresses are activated by regular keypress, and not by shorting(I thought... Very strange...).

Anyway, this should indicate that the problems are not at the keyswitches.

Now do the same exercise on the pins of the left rj45 socket. If those work, I would guess the cable is the culprit.

jcs021102, 2024-09-21:
I just changed to cable a second time and everything works! With the exception of the knob..
[image]

jcs021102, 2024-09-21:
I just found out the middle button and knob work, it just doesnt show in vial!!
Thanks for your time and patience man, you are the best

closed 2024-09-21 by jcs021102

tompi, 2024-09-21:
Great! The knob should register when you press it?

Turning it does not show up in vial, but should work anyway?