tompi/cheapino #71: Parts list expanded

djdanielsson, 2024-09-18:
I was looking for some other options that are currently not listed in your parts list that I thought might be nice to add for people like me who want for example a clicky keyboard or different keycaps. I know you can't list them all but I thought a few other options might be nice. I am happy to PR something if anyone has some suggestions. I am currently looking to find some clicky switches if anyone has found any that work well.

tompi, 2024-09-19:
Hey, sure, clickies are nice for home-office workers and sadists ;)

I have a cheapino with cherry blue switches which I really like, but my wife threatens to divorce me every time I use it...

Please add some more options in a pull request, Ill be happy to merge!

P.S. Im always on the lookout for nice blank keycaps, and also low profile mx, let me know if you come across some nice ones.

djdanielsson, 2024-09-19:
Do you have a link to the cherry blue switches that you have?

tompi, 2024-09-19:
Unfortunately no, harvested from old keyboard. But i assume they Are like any cherry blues...

SadWitchBoards, 2024-09-19:
Honestly at this point I feel like we need a Cheapino fan page to share builds and shopping links. 🤣 As for clicky switches, I feel the cheaper Kailh box clicky switches or even the Gateron Melodic are good examples now.

djdanielsson, 2024-09-19:
That's not a bad idea, maybe a Reddit thread for that which could be linked somewhere or similar.

SadWitchBoards, 2024-09-19:
Maybe a subreddit with resources for budget ergo builds and a master list for Cheapino links?

tompi, 2024-10-06:
Feel free to make a subreddit ;)

Closing this issue for now.

closed 2024-10-06 by tompi