tompi/cheapino #72: Request for MCU-Covered Case for Cheapino v2

azzamsa, 2024-09-19:
Hi 👋🏽,

I absolutely love everything about Cheapino, but one thing that bothers me is the exposed MCU. I’m a bit paranoid about solder residue, so I clean it frequently and use a full-coverage case for my keyboard.

Most of the time, I use [Chewiedies's case](https://www.printables.com/@chewiedies_360738/models), which is great because it only requires a 3D printer thanks to its gasket mount design. I love it, especially since sourcing exact screws or heat inserts is really difficult here. I use the case as-is, without any screws or magnets, as my main goal is simply to cover the MCU and PCB to prevent accidental contact.

Cheapino already has a v2 case, but the MCU is still exposed. How difficult would it be to add a version where the MCU is covered? I don’t have any 3D modeling experience beyond using Tinkercad to cover a TRRS hole.

Here’s an example of what I’m envisioning:

[image]

I am also surprised by its size, a whopping 24 MB.

[image]

Thanks a lot in advance 🙏🏽

tompi, 2024-09-19:
Hi! 

I thought about this, and there are several problems when doing this:

1. It will be very difficult to print if the top is not uniform(today it prints with top down, since top is even. if you wanted mcu cover, there would be no surface to be placed down on the bed, so maybe you would print it 45degrees angle with supports?
2. The height of the mcu varies drastically depending on which type of mount you choose(pin headers or sockets, and there are very different height sockets...)
3. The LED would be hidden. (P.T. its controlled by qmk, so you can use it for effects, caps lock, layer indication etc)

I think a solution with a separate mcu-"lid" would be best, some type of snap on...

I tried to use tinkercad myself, but very quickly ran into size limitations, and switched to openscad, which is VERY cool :)
You can find the openscad files in the repo if you like to try, otherwise I will keep this and use it for a lid-addon whenever I get some time...

SadWitchBoards, 2024-09-19:
Snap on or magnetic  with a height based on the 3.5 headers on the shopping list and a hole for the led to shine through, perhaps?

azzamsa, 2024-11-22:
Hi,  

I’ve been working on modifying the Cheapino case to add an MCU and an RJ45 cover. The cover might not be for everyone, but I wanted to share my journey and get some suggestions along the way.  

---

Having no prior experience with OpenSCAD, I initially jumped into TinkerCAD. However, I ran into issues with the case's large size, which is caused by the `$fn=50` variable in OpenSCAD.  

After removing `$fn`, I was able to work on the case in TinkerCAD without issues. But creating fillets in TinkerCAD turned out to be a real pain. So, I switched to FreeCAD. Unfortunately, modifying STL files in FreeCAD was painfully slow due to the huge mesh of the case. Simple operations often took forever, and saving the file frequently caused my system to hang. On several occasions, I even had to restart my machine because it became completely unresponsive—even though it’s a fairly powerful setup.  

I also tried Plasticity, but modifying existing STL files there wasn’t straightforward either.  

The OpenSCAD workbench in FreeCAD worked well for simple models but failed when handling the case. I brought up this issue several times on the FreeCAD Discord server, but received no response. An admin mentioned that the OpenSCAD workbench doesn’t get much attention, so bugs are to be expected.  

Next, I tried OnShape. While it handled large STL imports well, I wasn’t interested in investing time learning a tool I don’t enjoy.  

Finally, I decided to give OpenSCAD another shot. Initially, I manually adjusted positions, which was time-consuming. Eventually, I realized I could copy the positions from the original model. At first, I tried building the structure using three cubes (|-|), but later learned it’s better to use two cubes with `difference`.  

That said, there are still challenges with OpenSCAD. Rotating items in preview mode is cumbersome, as you have to render the model first before adjusting its position. To simplify this, I started using `$fn = $preview ? 10 : 50;`  

I turned to OpenSCAD because creating fillets in TinkerCAD was frustrating, but it’s proving to be a similar struggle here. I also attempted to use FreeCAD to add fillets to an exported STL file, but it just kept crashing.  

I don’t want to use `minkowski` for filleting because it messes up the dimensions by adding extra size to the cube. The `hull` function works well for the sides, but I’d need additional steps to fillet the top. Using an SVG imported into OpenSCAD could work, but I’d face the same issue for the top side.  

---

Hi @tompi,  

Would you mind reviewing my modifications before I send them for printing? Any suggestions—especially regarding the fillets—would be really helpful.  

Since I don’t have a 3D printer, I’ll need to send this out to get printed. I’ve already checked the dimensions using the ruler tool, and everything seems to match my requirements.  

For reference, I have a 13mm RJ45 (measured from the PCB) and a 9mm MCU (also from the PCB).  

Thanks in advance!  

---

[image]

```diff
From 1af1ddb71185924ef26874b7ab845e6ed8207fc6 Mon Sep 17 00:00:00 2001
From: "Azzam S.A" <noreply@azzamsa.com>
Date: Thu, 21 Nov 2024 04:53:49 +0700
Subject: [PATCH 1/3] feat: rj45 cover (right side)

---
 case/cheapino-top-right.scad | 26 +++++++++++++++++++++++++-
 1 file changed, 25 insertions(+), 1 deletion(-)

diff --git a/case/cheapino-top-right.scad b/case/cheapino-top-right.scad
index 1508715..14985dd 100644
--- a/case/cheapino-top-right.scad
+++ b/case/cheapino-top-right.scad
@@ -1,7 +1,7 @@
 height = 95.1738;
 width = 132.6134;
 step = 0.06;
-$fn=50;
+$fn = $preview ? 10 : 50;
 
 height_translation = 0;
 include <modules.scad>
@@ -26,10 +26,12 @@ mirror() {
 
     // RJ45 socket
     color("cyan") {
+        // main hole
         translate([29.2, -9.33, 4.1])
         linear_extrude(10)
[... 132 lines cut ...]
+  color("gray") {
+    difference() {
+      // x = left/right, y = front/back
+      translate([29.5 - 2, 10.2 - 2, 4.1])
+        // MCU+hotswap height is 9mm
+        // I set 11, then I will use a difference to get 9mm
+        linear_extrude(11)
+        // width = left/right, height = front/back
+        // Add thickness of 4. 2 left 2 right.
+        square([19.5 + 4, 25 + 2]);
+
+      // repeat the difference() for initial mcu cutout
+      translate([29.2, 10.2, 4.1])
+        linear_extrude(9)
+        square([19.5, 25]);
+    }
+}
+
 
 
 
-- 
2.47.0

```

SadWitchBoards, 2024-11-22:
I have a printer and can print this and share photos if you like.

azzamsa, 2024-11-23:
Thanks! 🙏🏻

But, let's wait for Tompi response. Maybe he got a better design or a suggestion for easier steps to do fillets.

tompi, 2024-11-23:
Looks good, I really like your $preview conditional $fn solution!

It would probably be nice with a bit rounded corners for the rj45/mcu covers?

If you make it a pull request, its much easier to cooperate on, if I want to make a suggestion, or if @SadWitchBoards  wants to test it, it will be super easy to just use the PR ?

SadWitchBoards, 2024-11-23:
Sounds good. Just send me any files you want tested and I will print them and try them out on one of my boards.

azzamsa, 2024-11-24:
> It would probably be nice with a bit rounded corners for the rj45/mcu covers?

> I don’t want to use minkowski for filleting because it messes up the dimensions by adding extra size to the cube. The hull function works well for the sides, but I’d need additional steps to fillet [...]

@tompi do you have any comments on a better way to do fillet in OpenSCAD?

tompi, 2024-11-24:
Thanks for the PR! 

For the rounded corners: I was thinking starting with 2D rounded rectangle, and then, with a loop, shrink it and copy it on top of the previous.

This is how the case is rounded on the case.