tompi/cheapino #74: Compatibility/support for Seeed Studio XIAO?

LargeCrowd, 2024-09-23:
Hi tompi! I came across the cheapino while searching for affordable split keyboards and I'm super excited to eventually build my own!

I was just wondering if the pcb has support for the Seeed Studio XIAO rp2040, because I saw a photo of someone's build using it (although I can't find it anymore). Given its smaller form factor, i thought it might look nicer on the cheapino! (I also saw that you're thinking of working on wireless compatibility using a XIAO for the v3?)

I'm also curious what other improvements you're planning for the v3 and when you plan to release it :)

SadWitchBoards, 2024-09-24:
I don't find Xiao that inexpensive compared to the generic Nice! Nano. I got 5 for the price of one Seed Xiao ble.

MatthieuTricoire, 2024-09-25:
> I don't find Xiao that inexpensive compared to the generic Nice! Nano. I got 5 for the price of one Seed Xiao ble.
But on the generic Nice Nano, there's a battery drain issue, isn't there?

tompi, 2024-09-25:
Hey, I actually have a cheapino v2 partially working with xiao ble and zmk, picture of it here:
https://github.com/tompi/cheapino/blob/master/doc/images/orderingguide/keycaps.png

Here is the current zmk status:
https://github.com/tompi/cheapinov2-zmk-config

I just need some time to finish the zmk profile, I just started doing other stuff...
I will probably pick it up again this fall, when outdoor activities start being less pleasant...

I tagged some issues with v3, mostly minor stuff...
Would really like to get rid of all ghosting(although its perfectly manageble in firmware, I dont like it...)

Have been toying with the idea of adding leds and sound, but not sure the added complexity is worth it...

There is also some demand for a choc version and a 6th column, but it also demands sacrifices(choc support would probably mean i have to choose either hotswap or soldered...)

v3 is probably not coming anytime soon...

SadWitchBoards, 2024-09-25:
> > I don't find Xiao that inexpensive compared to the generic Nice! Nano. I got 5 for the price of one Seed Xiao ble.
> 
> But on the generic Nice Nano, there's a battery drain issue, isn't there?

That was fixed in May and the current batches don't have that issue.

SadWitchBoards, 2024-09-25:
> Hey, I actually have a cheapino v2 partially working with xiao ble and zmk, picture of it here: https://github.com/tompi/cheapino/blob/master/doc/images/orderingguide/keycaps.png
>
> Here is the [...]

Sound: While neat, I think you're right about the complexity maybe not being worth it. I've got a few boards with this integrated and never use it and know others with the same feelings. 

LEDs: I don't personally care about RGB and I think it takes away from the simplicity of the Cheapino as a beautiful budget board. Personal opinion of course. 

Choc: what restricts this to solder? Could it possibly be compatible with choc V2? There has been a lot of movement in that version lately, including the best feeling and sounding low profile switches that I've used in the new all pom selections. Only difference is a larger Center hole in the PCB, which undoubtedly would cause a slightly larger PCB. Not sure if that's worth it to you. 

So many considerations. Thank you for taking time to go through it all.

tompi, 2024-09-25:
> Choc: what restricts this to solder? Could it possibly be compatible with choc V2? There has been a lot of movement in that version lately, including the best feeling and sounding low profile [...]

I was thinking I could maybe do some combination of choc(v2)/mx, but seems the center hole is different size? Then it would require completely differen pcb, and the case would alse be different... The routing would need to be completely redone and everything is just 2x the work...

azzamsa, 2024-09-26:
> Hey, I’ve got a partially working Cheapino v2 with Xiao BLE and ZMK! Here’s a picture of it:

Wow, I thought the PCB would need some modifications.

> I was considering a mix of Choc (v2) and MX switches, but it looks like the center hole is a different size. This would mean needing a completely new PCB and case, not to mention redoing all the routing. It’s basically twice the work…

I love the Cheapino for its simplicity, and I align with its "cheap" mission. Adding extra features that don’t enhance productivity feels counterproductive.

Personally, I never use LEDs, OLEDs, speakers, haptics, or wireless in any of my builds due to budget constraints, so the Cheapino fits my needs perfectly.

I think you might want to start using a `wontfix` tag or mention it in the README to avoid getting bombarded with feature requests that don’t align with your goals. Since it’s open source, people can always fork the project. You could even help them out in their own repo if needed.

You could merge forks that add things like LEDs or speakers into a `contrib/` or `mods/` directory, but it’ll add more work to maintain, and maintainers come and go.

I don’t plan to build a wireless keyboard anytime soon as I’m still searching for my "endgame" build. But if I do, and it involves the Cheapino, I’d likely fork it and reach out to you for advice if I plan to add more features.

azzamsa, 2024-09-26:
I think it’s best to keep Cheapino as it is. For a potential v3, maybe just add essential features like wireless.

Let Choc users fork it and give it a new name, like `cheapino-choc`, and the same goes for users who want a 6-column version.

LargeCrowd, 2024-09-27:
> I love the Cheapino for its simplicity, and I align with its "cheap" mission. Adding extra features that don’t enhance productivity feels counterproductive.
>
>
>
> Personally, I never use LEDs, OLEDs, [...]

I agree. It has never occurred to me that these features are important to my typing experience (although, to be fair, I wouldn't consider myself to be that deep into the keyboard community at all).

I'd only just found this split ergo keyboard community recently and thought that it seemed niche and interesting!

Since I found the Cheapino, I've come across a few other split keyboard builds (mainly from Seeed's Xiao keyboard contest held a couple of years back) and personally, I found it really interesting how everyone implemented their own preferences into their designs, making all the keyboards different!

@tompi I appreciate your hard work and I hope you continue to find joy in it as you work on the v3. No pressure to complete it, I look forward to find out what's in store!

glennmca-dev, 2024-10-01:
RE lights: i agree they may add unnecessary cost. but in the spirit of the cheapino, you could go the route of adding 3-5 optional underglow LEDs per side so only the bottom has LEDs.
I mentioned this on https://github.com/tompi/cheapino/issues/80 aswell, but if wireless, is it possible to use the empty RJ45 space for two encoders?

tompi, 2024-10-05:
@LargeCrowd Im gonna close this now, I dont think I will put in the effort to port to seed studio xiao rp2040.
I know it was very popular for a while, but its more than double the price of rp2040-zero, so does not seem like a good deal anymore.

That said, the cheapino and qmk should support it. If you want to try, make sure you mount it at the "back" of the pin holes, leaving the 2 pin holes closest to the edge empty. Should just be a matter of changing the mcu in qmk and changing the pin-definitions after that...

closed 2024-10-05 by tompi