tompi/cheapino #77: matrix_row_t unknown type name in encoder.c

CannonFodderSE, 2024-09-29:
Greetings,
When I try to compile the Cheapinov2 in QMK I get the error below:

`Compiling: drivers/encoder/encoder_quadrature.c                                                    In file included from drivers/encoder/encoder_quadrature.c:6: keyboards/cannonfodder/encoder.h:5:25: error: unknown type name 'matrix_row_t' 5 | void fix_encoder_action(matrix_row_t current_matrix[]); `

'matrix_row_t' is used encoder.c, ghosting.c, ghosting.h and matrix.c without throwing an error.

Do you have any idea why this is happening or how to correct it?

tompi, 2024-09-30:
Hey. From the error it seems like you did some modding? It should compile if you follow the firmware guide step by step?

CannonFodderSE, 2024-09-30:
Greetings,
Tried to compile the stock code from [https://github.com/tompi/vial-qmk/tree/cheapinov2](https://github.com/tompi/vial-qmk/tree/cheapinov2) and I get the error below for the default keyboard.

`Compiling: keyboards/cheapino/encoder.c                                                            keyboards/cheapino/encoder.c: In function 'turned':
keyboards/cheapino/encoder.c:28:5: error: implicit declaration of function 'encoder_exec_mapping' [-Werror=implicit-function-declaration]
   28 |     encoder_exec_mapping(0, clockwise);
`

When I try Vial keyboard I get the errors below:

`Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/quantum.h:210,
                 from ./.build/obj_cheapino_vial/src/default_keyboard.h:27,
                 from ./keyboards/cheapino/keymaps/vial/keymap.c:1,
                 from quantum/keymap_introspection.c:5:
quantum/encoder.h:115:13: error: excess elements in array initializer [-Werror]
  115 |             { (cw), (ccw) }
      |             ^
./keyboards/cheapino/keymaps/vial/keymap.c:17:15: note: in expansion of macro 'ENCODER_CCW_CW'
   17 |     [0] =   { ENCODER_CCW_CW(KC_MS_WH_UP, KC_MS_WH_DOWN)},
      |               ^~~~~~~~~~~~~~
quantum/encoder.h:115:13: note: (near initialization for 'encoder_map[0]')
  115 |             { (cw), (ccw) }
      |             ^
./keyboards/cheapino/keymaps/vial/keymap.c:17:15: note: in expansion of macro 'ENCODER_CCW_CW'
   17 |     [0] =   { ENCODER_CCW_CW(KC_MS_WH_UP, KC_MS_WH_DOWN)},
      |               ^~~~~~~~~~~~~~
quantum/encoder.h:115:13: error: excess elements in array initializer [-Werror]
  115 |             { (cw), (ccw) }
      |             ^
./keyboards/cheapino/keymaps/vial/keymap.c:18:15: note: in expansion of macro 'ENCODER_CCW_CW'
   18 |     [1] =   { ENCODER_CCW_CW(RGB_HUD, RGB_HUI)  },
      |               ^~~~~~~~~~~~~~
quantum/encoder.h:115:13: note: (near initialization for 'encoder_map[1]')
  115 |             { (cw), (ccw) }
      |             ^
./keyboards/cheapino/keymaps/vial/keymap.c:18:15: note: in expansion of macro 'ENCODER_CCW_CW'
   18 |     [1] =   { ENCODER_CCW_CW(RGB_HUD, RGB_HUI)  },
      |               ^~~~~~~~~~~~~~
quantum/encoder.h:115:13: error: excess elements in array initializer [-Werror]
  115 |             { (cw), (ccw) }
      |             ^
./keyboards/cheapino/keymaps/vial/keymap.c:19:15: note: in expansion of macro 'ENCODER_CCW_CW'
   19 |     [2] =   { ENCODER_CCW_CW(RGB_VAD, RGB_VAI)  },
      |               ^~~~~~~~~~~~~~
quantum/encoder.h:115:13: note: (near initialization for 'encoder_map[2]')
  115 |             { (cw), (ccw) }
      |             ^
./keyboards/cheapino/keymaps/vial/keymap.c:19:15: note: in expansion of macro 'ENCODER_CCW_CW'
   19 |     [2] =   { ENCODER_CCW_CW(RGB_VAD, RGB_VAI)  },
      |               ^~~~~~~~~~~~~~
quantum/encoder.h:115:13: error: excess elements in array initializer [-Werror]
  115 |             { (cw), (ccw) }
      |             ^
./keyboards/cheapino/keymaps/vial/keymap.c:20:15: note: in expansion of macro 'ENCODER_CCW_CW'
   20 |     [3] =   { ENCODER_CCW_CW(RGB_RMOD, RGB_MOD) },
      |               ^~~~~~~~~~~~~~
quantum/encoder.h:115:13: note: (near initialization for 'encoder_map[3]')
  115 |             { (cw), (ccw) }
      |             ^
./keyboards/cheapino/keymaps/vial/keymap.c:20:15: note: in expansion of macro 'ENCODER_CCW_CW'
   20 |     [3] =   { ENCODER_CCW_CW(RGB_RMOD, RGB_MOD) },
      |               ^~~~~~~~~~~~~~
quantum/encoder.h:115:13: error: excess elements in array initializer [-Werror]
  115 |             { (cw), (ccw) }
      |             ^
./keyboards/cheapino/keymaps/vial/keymap.c:21:15: note: in expansion of macro 'ENCODER_CCW_CW'
   21 |     [4] =   { ENCODER_CCW_CW(KC_NO,KC_NO)  },
      |               ^~~~~~~~~~~~~~
quantum/encoder.h:115:13: note: (near initialization for 'encoder_map[4]')
  115 |             { (cw), (ccw) }
[... 55 lines cut ...]
   60 | _Static_assert(NUM_KEYMAP_LAYERS_RAW == NUM_ENCODERMAP_LAYERS_RAW, "Number of encoder_map layers doesn't match the number of keymap layers");
      |                                      ^
quantum/keymap_introspection.c: In function 'keycode_at_encodermap_location_raw':
quantum/keymap_introspection.c:50:70: error: division by zero [-Werror=div-by-zero]
   50 | #    define NUM_ENCODERMAP_LAYERS_RAW ((uint8_t)(sizeof(encoder_map) / ((NUM_ENCODERS) * (NUM_DIRECTIONS) * sizeof(uint16_t))))
      |                                                                      ^
quantum/keymap_introspection.c:63:21: note: in expansion of macro 'NUM_ENCODERMAP_LAYERS_RAW'
   63 |     if (layer_num < NUM_ENCODERMAP_LAYERS_RAW && encoder_idx < NUM_ENCODERS) {
`

When I try tompi keyboard I get the below error for each layer:

`Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/keymap_introspection.c:5:
./.build/obj_cheapino_tompi/src/keymap.c:13:380: error: macro "LAYOUT_split_3x5_3" requires 37 arguments, but only 36 given
   13 | T(KC_E), LALT_T(KC_I), LGUI_T(KC_O), KC_Z, KC_X, KC_C, KC_D, LT(5,KC_V), LT(1,KC_K), KC_H, KC_COMM, KC_DOT, KC_SLSH, LT(6,KC_ESC), LT(3,KC_SPC), LT(5,KC_TAB), LT(1,KC_DEL), LT(2,KC_BSPC), LT(4,KC_ENT)),`

So I started with a blank keyboard and started importing the matrix then ghost features.  They work.  Modified the RGB for capslock. 

[... cut ...]
All of this came about because I wanted to have the LED change color for capslock and a different color for a seperate default layer :)

As side note, the RGB LED on the RP2040-Zero I'm using have the red and green diodes swapped.  Don't know if this is standard for them or I got  a nonstandard module.  Took me for ever to figure it out trying to set the color using HSV.  The colors were always wrong.  When I switched to setting it by RGB I finally figured it out.

Love the keyboard by the way.

tompi, 2024-09-30:
The changes in quantom/encoder.c and .h files.

tompi, 2024-09-30:
P.S. In regular qmk I just called a user function, but in Vial, this was no good and you could not control encoder from vial, hence the hack...

CannonFodderSE, 2024-09-30:
Looks like QMK did a major overhaul of the encoder function. `encoder_exec_mapping` is no long in quantum/encoder.c.  It is now 297 line vice 179.  Look like I'll have to figure out what the new equivalent function is.

tompi, 2024-10-12:
Hey, just tested myself, and it compiled fine with the existing branch, right? Are you trying to get it up to latest qmk version? (cause that might be difficult with vial i think...)

This is what I dit:

```sh
git clone git@github.com:tompi/vial-qmk.git qmk_test
cd qmk_test
make git-submodules
qmk compile -kb cheapino -km vial
```

CannonFodderSE, 2024-10-12:
Thomas,
Yes, I'm trying with the latest QMK.  I did the RGB and encoder from scratch.  I works with the standard QMK build.

Got issues with the vial build to work.  Been a bit busy with so I haven't had time to work on it.

When I get a chance I'll try your instructions above.

Thanks for a great design.  I just have to get my muscle memory to adjust to the new setup.

Have great  day,
Randy

tompi, 2024-12-29:
I assume problem was trying to merge latest qmk into current vial(which wont work, vial is locked to older qmk)

closed 2024-12-29 by tompi