tompi/cheapino #78: MacOS compatibility

ProcedurallyGeneratedGoldblum, 2024-09-29:
Has anyone successfully got the Rp2040-Nano Cheapino to work with MacOS? I just cannot get the board to even turn on, unless I'm missing something.

soul-wish, 2024-09-30:
I have two of Cheapino's and I use them with MacOS only. Have no issues with them. I'm using Macbook Pro M1. So, I'm not sure what exact issue you may encounter.

tompi, 2024-09-30:
Hi! Send some pictures and a description of what you tried. 

Where you able to Flash the mcu?

Ps: i assume you mean rp2040-zero. The rp2040-nano is pretty big and i doubt it would Even fit on the cheapino...

ProcedurallyGeneratedGoldblum, 2024-09-30:
Apologies, I should have been more specific. The rp2040-zero, yes. It works fine with my Windows machine. The Cheapino v2 is fully built but only works with my Windows machine. When I connect to my 2016 Macbook Pro, it does not even detect the MCU is there (the LED does not illuminate).

Just wondering if yourself or anyone else has any feedback before I deep dive.

Thanks!

tompi, 2024-09-30:
This is probably a feature of the mac: Im guessing the docking is not approved? (it should ask permission when you connect)

Otherwise it could be a cable issue(are you using the same cable?)

P.S.: The keyboard functions exactly the same on windows and mac, there is no difference...

djdanielsson, 2024-09-30:
I have gotten it to work on MacOS, Windows, and Linux. I could pull out my really old Mac to see if I have an issue with that but my M1 works fine.

ProcedurallyGeneratedGoldblum, 2024-09-30:
After going back and forth about this...I now have three separate cables that were tested in succession...the fourth allowed for the connection. 

Thanks so much! :)

tompi, 2024-10-02:
Super! I assume the cable situation is the same for windows? It should not work with one cable on windows and not work with the same cable on mac?

ProcedurallyGeneratedGoldblum, 2024-10-04:
It was quite confusing. All of the cables are USB-C to USB-C rated for both data and power. 3 out of the 4 worked on the Windows machine where the 4th worked for both. Not quite sure how to interpret that one! 3 crappy cables?

tompi, 2024-10-04:
Just checking: did you plug all of the 4 cables directly into the Mac?

Sorry to bother you but could you snap a picture of the cables that didnt work?

ProcedurallyGeneratedGoldblum, 2024-10-05:
Ill snap pictures when I can but for the moment;

Cable1: Apple USB-C -USB-C (Windows connection only)
Cable2: Samsung phone USB-C - USB-C (Windows connection only)
Cable3: Apple USB-A (M) - USB-C **<-----------successful connection with Windows machine / MBP**
Cable4: Random USB-A (M) - USB-C (Unsure of spec (Windows connection only))

tompi, 2024-10-06:
Just linking to related reddit post: 
https://www.reddit.com/r/ErgoMechKeyboards/s/h1N7jalBm9

Starting to suspect nkro might act up in rare occasions.
@ProcedurallyGeneratedGoldblum : you are not using a kvm or usb hub? You tested connected directly to machine?

tompi, 2024-10-06:
Just testing some more, and I can reproduce the problem with a high-end cable on my m1, but the same cable works fine on my older intel mac(2019),and also works fine on my ipad... crazy stuff, starting to suspect hardware issue with silicon platform??? Is your mac m1/2/3?

ProcedurallyGeneratedGoldblum, 2024-10-06:
The connections were made directly to both machines, no hubs or KVMs were used.

The MacBook Pro is 2016, the intel powered machine.

tompi, 2024-10-06:
Yep, I agree, after chatting with tweetiedabirdie (which knows his stuff), seems like this is an issue with all custom keyboards...(I verified that the same think is happening with my skeletyl...).

I added info to trouble shooting and ordering guide, will close the issue.
P.S.: Moral of the story is to use cheaper cables ;)

closed 2024-10-06 by tompi