tompi/cheapino #80: Wireless for V3?

glennmca-dev, 2024-09-30:
I'm not sure if there's anything i can do to help or what the limitations are if this has been discussed before, but a wireless version using a nice!nano or NRF52840 board would be amazing (halves still wired but wireless to the host device)

azzamsa, 2024-10-01:
You can build a wireless version using the v2 (current version) using Seeed Xiao, See https://github.com/tompi/cheapino/issues/74#issuecomment-2373630023

glennmca-dev, 2024-10-01:
Ahh awesome. I already have my V2 working well so i may wait it out for the V3 with the Seeed Xiao fully supported. I couldn't be without my rotary encoder. Maybe V3 if wireless could replace the RJ45 space with two encoders

closed 2024-10-01 by glennmca-dev