tompi/cheapino #81: Confused on starter keymap layout

cgregurich, 2024-10-01:
Just built my cheapino and the default layout/layers you have in the vial website (?) seems to be kind of wacky. Eg. no shift, ctrl, backspace, etc. I know that it's super customizable but because this is my first ergo keyboard, I'm not sure where to even begin so I need a good starting layout that more or less gets the job done "out of the box". Can anyone help me with this?

tompi, 2024-10-01:
Yeah, didnt put much effort into the vial default keymapping... there is a miryoku firmware, which is kind of the goto standard...

Maybe Flash that and give it a go?

cgregurich, 2024-10-02:
I'll try that, thanks for the quick reply :)

tompi, 2024-11-10:
Closing, default keymap is mostly for testing. Ppl will eithe compile themselves or use vial or miryoku.

closed 2024-11-10 by tompi